ARM-software/tf-issues #286: Fix aff_state inconsistency for SUSPEND states

soby-mathew, 2015-01-14:
In the current implementation of node state management in PSCI, when a core is suspended, the state of that core will be reported as ON by PSCI_AFFINITY_INFO. But if that core is the last one within the cluster, then the state of the cluster will be reported as OFF while that of the core will be reported as ON which is a inconsistency.

This fix would involve reworking how the ref counts are managed within PSCI.

vwadekar, 2015-01-15:
IMO, this inconsistency would be seen for all cores when th cluster enters
suspend state. Do you think marking a core as OFF when it enters suspend
would be better? I think when a cluster enters suspend, all the cores in
that cluster should also be in the same state. Since the PSCI spec does not
have a Suspend state for a cluster/core, OFF is the only available option.

athoelke, 2015-01-15:
There are two related non-compliances in the current implementation of `AFFINITY_INFO` when targeting an affinity level above 0, i.e. for cluster or system state, when compared to the v0.2 specification.

The spec says that `AFFINITY_INFO` should report `ON` for cores (level 0) that are running or in a low power suspend state; only report if the core is `OFF` if it never been turned on or has called `CPU_OFF`; and only report `ON_PENDING` if `CPU_ON` has been called but the core has not completed warm boot in the firmware. This is implemented correctly.

For clusters or above, the current implementation reports the physical state of the affinity level: `ON` if at least one core (within the cluster) is running or suspended to an affinity level lower than the target of the `AFFINITY_INFO` call, otherwise it reports `OFF`. The special cases of no-cores-running-and-at-least-one-suspended and all-cores-OFF-or-ON_PENDING which should report `ON` and `ON_PENDING` are not reported as required in the specification.

The uses for these corner cases for affinity level 1 and above are not high priority and the cost of compiliance is non-trivial - please let us know if this is something that you need to have fixed.

vwadekar, 2015-01-16:
Thanks Andrew. I noticed that there is no CPU suspend state in the spec, so
the current implementation looks OK. The inconsistencies are not high
priority but would be good to have.

soby-mathew, 2015-10-07:
The PSCI 1.0 specification allows AFFINITY_INFO to return error for power levels higher than 0. The new PSCI framework in Trusted Firmware implements this behaviour. Hence this issue is no longer valid for Trusted Firmware.

closed 2015-10-07 by soby-mathew