ARM-software/tf-issues #290: how to get  root of trusted  key stored in the juno  platform?

xlyu, 2015-01-30:
I got tag v1.1-rc0 had achieved Trusted Borad Boot functions,including BL2,BL3x check.
However, how to get ROT keys from  juno platform is in todo list.
I have a juno board and DS-5 tools, Could anybody give me a simple guide how to set my own hash of root public key to APB mapped register(Table 2-12, Juno Soc Technical Reference Manual version r0p0)? and how to read from these APB mapped registers?
I'm new in  juno, Thanks.

danh-arm, 2015-01-30:
The APB mapped register bank representing the ROTPK is read only. I'm not able to share the values with you at this time. As you've observed, the current prototype TBB functionality doesn't use these registers yet but they will be used in future.

Incidentally, I'm not sure which table you're referring to as there is no Table 2-12 in the version of the Juno SoC TRM I'm looking at (Issue C)

xlyu, 2015-02-04:
Sorry to reply so late to you.thanks very much for your answer.
I just wonder which boot step to set ROTPK/EK.. values to  these APB mapped register bank, is it BL30 or anything else,  and how to set? if it's not allowed, it will be useless to do secure boot in juno development board, because of lack of root of trusted private key, we can't sign BL2,BL3x image.
meanwhile, secure boot need  a NV monotonic counter to avoid replay hacking, It also need read and write to  juno NV region that the APB mapped.
Any insight or help is greatly appreciated.
we signed the arm NDA,and now just wait a  assigned account  to get PDD through Dropzone, if we can get Juno SoC TRM Issue C document after joining,because we just only can download Issue B on the arm website currently.

athoelke, 2015-02-06:
This register bank is once-programmable, read-only. It has been programmed with a root public key for Juno and you cannot replace the value with your own key.

However, we are not using the ROM in Juno to boot the board as the BL1 image used to boot Juno is programmed into Flash. The BL1 image for Juno in this release *does not* verify that the keys used to sign the BL2, TrustedWorld and NonTrustedWorld certificates is the ROTPK in the Juno registers. See [`plat_match_rotpk()`](ARM-software/arm-trusted-firmware/blob/master/plat/juno/juno_trusted_boot.c#L41) in the Juno platform port.

It is possible to embed the ROTPK (or hash of it) into BL1 and verify that the signing keys in these certificates matches the embedded key by modifying the current implementation of `plat_match_rotpk()`. This would anchor the chain of trust in the BL1 image used to boot the Juno board.

But clearly it is not possible to fully secure the boot on Juno as it is possible to bypass the ROM and execute an arbitrary BL1 image that has been loaded into the flash memory.

xlyu, 2015-02-10:
This is very detailed information of what i'm looking for. thanks danh and athoelke 
In Juno board, except the BL1 flashed into the flash memory. the another import is ROTPK can't  be used in real.It just show how secure boot can be processed. Looking for the root of trust private key  matched ROTPK which stored in juno board is incorrect.

closed 2015-02-18 by danh-arm