ARM-software/tf-issues #298: Add memory barrier for mmio_write/mmio_read

Leo-Yan, 2015-03-18:
hi,

When i debug for hikey board, found after add memory barrier for mmio_write_[8|32|64], then ATF will be much stable. So just wander, for these related operations, ATF can provide some kind of APIs like below? This will be easily choosed by SoC vendors:

mmio_write_relax()
mmio_write()
mmio_write_raw()

danh-arm, 2015-06-08:
Sorry, I've just noticed this hasn't had a response. Although adding barriers after every write might make the system more stable, there would be a significant performance cost to this. I don't think the proposed new functions would add much since SoC vendors would still have make a decision to which flavour to use, in the same way as they have to currently have to understand when to use memory barriers. It also might encourage lazy programming - i.e. always using the flavour with a barrier "just in case".

So I would say no change is needed.

Leo-Yan, 2015-06-09:
> Sorry, I've just noticed this hasn't had a response. Although adding barriers after every write might make the system more stable, there would be a significant performance cost to this. I don't [...]

Hi Dan,

Thanks for reviewing.

This requirement is coming from Hikey board, the mainly reason
is there have some registers are placed into difference IPs but
they have some sequential dependency. This is very common within
Hi6220's design.

So i personally think ATF can provide variants APIs so that SoC
vendors can select best API. Anyway, your explain is also reasonable
for me, we can hold on until we have obvious requirement.

Thanks,
Leo Yan

athoelke, 2015-06-09:
The ARMv8 ARM is very clear about the memory ordering for Device memory types. In particular ordering is guaranteed as long as the accesses are within a small memory region (typically at least one page but maybe as much as 1MB) - but accesses to two different IP blocks will generally not be ordered.

As the need for ordering between different device accesses is less common, we would recommend that explicit dmb/dsb() calls are made where necessary to ensure correct behaviour and to make it clear in the code where such synchronisation/ordering is essential. Adding barriers to the helper functions will result in developers defaulting to use these "safe" versions which will have the substantial negative impact on performance that Dan highlighted.

closed 2015-10-07 by danh-arm