ARM-software/tf-issues #299: Slight BL32_LIMIT bug?

vchong, 2015-03-22:
Hi,

For Juno
if (PLAT_TSP_LOCATION_ID == PLAT_DRAM_ID) 
BL32_LIMIT = (DRAM_SEC_BASE + DRAM_SEC_SIZE - DRAM_SCP_SIZE) = 0xFFC00000
but according to https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/firmware-design.md
BL32_LIMIT = 0xFFE00000
so perhaps BL32_LIMIT = (DRAM_SEC_BASE + DRAM_SEC_SIZE)?

The dram block for bl32 doesn't really overlap that of the scp so there's really no need to minus the DRAM_SCP_SIZE, right?

vchong, 2015-03-22:
Along similar lines, TSP_SEC_MEM_SIZE should just be DRAM_SEC_SIZE instead of (DRAM_SEC_SIZE - DRAM_SCP_SIZE)?

If there's no bug, then what's the 2MB gap in between BL32_LIMIT and DRAM_SCP_BASE for?

Again stressing this is just the case for dram store.

Thanks!

danh-arm, 2015-03-23:
Hi

You're right - this is an inconsistency in the Juno platform port, although it shouldn't cause any issues in practice.

The ARM platform ports are currently undergoing a major refactoring process. This inconsistency will be addressed as part of that.

Thanks for pointing it out.

Dan.

danh-arm, 2015-06-08:
Should now be fixed in upstream code

closed 2015-06-08 by danh-arm