ARM-software/tf-issues #303: Add s/w emulation for cpu power cycle

Leo-Yan, 2015-04-22:
hi,

Sometimes we need use s/w emulation for cpu power cycle, so that when the cpu enter wfi, it will not really be powered off.

For this case, s/w can let the cpu exit from cache coherency domain and jump to the secure entry point; so just like the cpu has the same behaviour w/t reset by power controller.

This is a debug functionality, and can be used by code preparation phase, for FPGA debugging, and for bring up new typeout SoCs.

U can refer the code from: https://github.com/Leo-Yan/arm-trusted-firmware/commit/e633333a6d0b9034921290870f836bb3ce50d27c;

achingupta, 2015-04-22:
Hi Leo,

I agree with the idea of an emulated reset in principal. I think Linux does something similar using kexec. It is something we have thought about in the past. It could be a very useful to support detection of pending interrupts prior to the final wfi in the OFF or SUSPEND paths. In this case it makes sense to use a lightweight reset mechanism instead of letting the power controller do the usual. However, a proper solution will have platform specific dependencies and we have not found the time to create the right abstractions.

The simplest way of resetting a cpu would be through the use of the RMR_EL3 register. This will ensure that the hardware takes care of resetting the system register and cache state. There are architectural and platform dependencies here. This register is only available if the processor supports both the aarch64 and aarch32 modes in EL3. If the register is available then the platform's power controller still has to support the reset requests from the processor. 

In the absence of this feature, software should do at least the following (not sequenced):

1. Disable and invalidate the Instruction cache
2. Disable the MMU and clear the associated registers e.g. TTBRs, MAIRs etc. There is a massive assumption here that the platform is using identity mapped tables or at least the code which will disable the MMU is running from identity mapped memory.
3. Invalidate the TLB contents

If there is a need to wait for an interrupt then we also need to ensure that the GIC is able to forward interrupts to the processor. During normal execution this is always the case so we just need to ensure that the platform hook to suspend or turn off a cpu does not disable this mechanism. The platform hook should not program the power controller to turn off the cpu either.

Keeping in mind the above, the emulated reset will be used in the following two scenarios.

1. To detect a pending interrupt late in the power down sequence. In this case final wfi will be skipped and the emulated reset will be used to jump to the psci entrypoint. 
2. To pipe clean the power down and power up path in PSCI without a dependency on the platform power controller. In this case, the final wfi is needed. My point about being able to forward interrupts from the GIC applies as well.

AFAIU, what you are proposing falls under scenario 2. above. Please correct me if that is not the case. If so, then in your patch

1. There is no need to reprogram the GIC to enable forwarding of interrupts to the processor. It should not have been disabled in the first place by the platform suspend/off hook.
2. There is no need to maintain another mailbox data structure (psci_mailbox_t). After the emulated reset there could be a jump to psci_aff_on_finish_entry() or psci_aff_suspend_finish_entry() depending upon the context.
3. There is a need to tear down the cpu's architectural context but that should be after the wfi instead of being before it.

Before we go any further I would like to know if the above makes sense to you. Please let me know if I am missing anything.

Thanks,
Achin

Leo-Yan, 2015-04-23:
hi Achin,

i try to answer your questions in below inline.

> I agree with the idea of an emulated reset in principal. I think Linux does something similar using kexec. It is something we have thought about in the past. It could be a very useful to support [...]

i'm not familiar w/t kexec, the idea is mainly coming from cpu_reset
function in kernel.

> The simplest way of resetting a cpu would be through the use of the RMR_EL3 register. This will ensure that the hardware takes care of resetting the system register and cache state. There are [...]

Good to know this.

> In the absence of this feature, software should do at least the following (not sequenced):
>
> 1. Disable and invalidate the Instruction cache
> 2. Disable the MMU and clear the associated registers e.g. [...]

If need really power off the cpu by power controller, we need mask
nIRQCPU/nFIQCPU signals to cpu; and power controller should detect
nIRQOUT/nFIQOUT signals to wake up the cpu. s/w can disable GIC's cpu
IF to coporate w/t power controller to do this.

If cpu just stay in WFI state and will be waken up by IRQs, then s/w
cannot disable GIC's CPU IF; otherwise the signals to CPU will be
masked.

We can leave platform hook functions to handle this.

> Keeping in mind the above, the emulated reset will be used in the following two scenarios.
>
> 1. To detect a pending interrupt late in the power down sequence. In this case final wfi will be skipped [...]

Exactly, now just for scenario 2.

> 1. There is no need to reprogram the GIC to enable forwarding of interrupts to the processor. It should not have been disabled in the first place by the platform suspend/off hook.

After dumpping the register, GICC_CTLR will disable group 1's
interrupt; so if there have non-secure interrupt, it will be masked
and cpu cannot be waken up.

is this related w/t interrupt route setting in BL31?

> 2. There is no need to maintain another mailbox data structure (psci_mailbox_t). After the emulated reset there could be a jump to psci_aff_on_finish_entry() or psci_aff_suspend_finish_entry() depending upon the context.

if w/o mailbox, it will be safe for cpu suspend/resume flow. But for
hotplug, i have concern if there have spurious interrupt for the cpu,
in this case the cpu need a way to know if it need run into WFI again.

> 3. There is a need to tear down the cpu's architectural context but that should be after the wfi instead of being before it.

Correct; here should tear down cpu after wfi.

achingupta, 2015-04-26:
Hi Leo,

Apologies for the delay! Please see my response inline.

> hi Achin,
>
> i try to answer your questions in below inline.
>
> > I agree with the idea of an emulated reset in principal. I think Linux does something similar using kexec. It is something we have thought about in the past. It could be a very useful to support detection of pending\
 interrupts prior to the final wfi in the OFF or SUSPEND paths. In this case it makes sense to use a lightweight reset mechanism instead of letting the power controller do the usual. However, a proper solution will have\
 platform specific dependencies and we have not found the time to create the right abstractions.
>
> i'm not familiar w/t kexec, the idea is mainly coming from cpu_reset
> function in kernel.

I have not looked at the code in detail either. Kexec might converge to
cpu_reset finally.

>
> > The simplest way of resetting a cpu would be through the use of the RMR_EL3 register. This will ensure that the hardware takes care of resetting the system register and cache state. There are architectural and platf\
orm dependencies here. This register is only available if the processor supports both the aarch64 and aarch32 modes in EL3. If the register is available then the platform's power controller still has to support the rese\
t requests from the processor.
> Good to know this.
nning from identity mapped memory.
> > 3. Invalidate the TLB contents
> >
> > If there is a need to wait for an interrupt then we also need to ensure that the GIC is able to forward interrupts to the processor. During normal execution this is always the case so we just need to ensure that the\
 platform hook to suspend or turn off a cpu does not disable this mechanism. The platform hook should not program the power controller to turn off the cpu either.
> If need really power off the cpu by power controller, we need mask
> nIRQCPU/nFIQCPU signals to cpu; and power controller should detect
> nIRQOUT/nFIQOUT signals to wake up the cpu. s/w can disable [...]

Yes! AFAICS, The hooks to suspend or turn off an affinity instance should make
this decision.

>
plies as well.
> Exactly, now just for scenario 2.
>
>
> After dumpping the register, GICC_CTLR will disable group 1's
> interrupt; so if there have non-secure interrupt, it will be masked
> and cpu cannot be waken up.
>
> is [...]

I don't think so. The interrupt framework controls whether an interrupt is
routed to EL3 or a lower exception level. Is your platform PM hook disabling
Group 1 interrupts by any chance? I would expect the OS to disable group 1
interrupts at the GIC CPU interface before calling CPU_OFF/CPU_SUSPEND. Is that
not the case at your end?

>
> > 2. There is no need to maintain another mailbox data structure (psci_mailbox_t). After the emulated reset there could be a jump to psci_aff_on_finish_entry() or psci_aff_suspend_finish_entry() depending upon the con\
text.
>
> if w/o mailbox, it will be safe for cpu suspend/resume flow. But for
> hotplug, i have concern if there have spurious interrupt for the cpu,
> in this case the cpu need a way to know if it need run into WFI again.

In the case of CPU hotplug, all interrupts (except SGIs) should be migrated away
from a CPU which is being turned off. So a spurious interrupt should not get
trigerred unless it is an SGI meant to wake the CPU up. Is that not the case on
the Hikey Board?

Thanks a lot!
Achin

> Correct; here should tear down cpu after wfi.
>
>
> �

-- IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium.  Thank you.

ARM Limited, Registered office 110 Fulbourn Road, Cambridge CB1 9NJ, Registered in England & Wales, Company No:  2557590
ARM Holdings plc, Registered office 110 Fulbourn Road, Cambridge CB1 9NJ, Registered in England & Wales, Company No:  2548782

Leo-Yan, 2015-04-27:
On Sun, Apr 26, 2015 at 01:54:38PM -0700, achingupta wrote:
> Hi Leo,
> 
> Apologies for the delay! Please see my response inline.

No matter ;)

> On Thu, Apr 23, 2015 at 07:55:49AM +0100, Leo-Yan wrote:
> > On Wed, Apr 22, 2015 at 02:24:57AM -0700, achingupta wrote:

[...]

> plies as well.
>
> I don't think so. The interrupt framework controls whether an interrupt is
> routed to EL3 or a lower exception level. Is your platform PM hook disabling
> Group 1 interrupts by any [...]

Thanks for clarification, i will debug further more for this issue,
and check where have broken this assumption.

> text.
>
> In the case of CPU hotplug, all interrupts (except SGIs) should be migrated away
> from a CPU which is being turned off. So a spurious interrupt should not get
> trigerred unless it is an SGI [...]

Understand, i will do some testing like this way on Hikey and let u
know the result.

Thanks,
Leo Yan