ARM-software/tf-issues #309: ATF support for ARMv7-A

sammoser, 2015-06-09:
Is there any interest in contributions to support AArch32 or ARMv7-A?
Has anyone done any work in this area or have any recommendations on how to go about it?

In slide 15 from this presentation from LCA14 it says there are no barriers to porting ATF to ARMv7-A.
http://www.slideshare.net/linaroorg/lca14-102-adoptingarmtrustedfirmware

athoelke, 2015-06-09:
Do you have something in progress that you are thinking of sharing, or are you just at the stage of wondering if it is worth the attempt?

As I said last year: technically the structure and design of Trusted Firmware can be ported to run on ARMv7-A. That was not to say it would be a simple update if you want to make it a good fit for the ARMv7-A exception architecture.

There are some technical issues to consider:
*  the ARMv7-A architecture does not make a clean division between monitor mode and secure svc mode: in particular they share a single set of memory translation tables. As a result it is not easy to divide responsibility between BL3-1 (the Monitor code) and BL3-2 (the Secure Payload/Trusted OS) in the way this is done in ARM Trusted Firmware for ARMv8-A.
* the system boots into secure svc mode. The placement of the BL1 and BL2 stages into execution modes and the transitions between them would need to be rework and handled differently for ARMv7-A
* the cost of switching modes is very different, which may influence the placement of services within the secure firmware. On ARMv7-A a transition from non-secure modes into monitor mode (and back again) is the dominant cost of world switching as this will result in switching between the non-secure and secure versions of the system registers. Switching from monitor mode to other secure modes is comparatively cheap. On ARMv8-A the system registers are switched in software, and only done if it is necessary to switch all the way to Secure-EL1.

There is also less demand from the ecosystem:
* PSCI has limited traction in ARMv7-A because the Linux kernel does not require it for this architecture
* the coupling of monitor and Trusted OS in the architecture means that even with Trusted Firmware, there would always be a significant integration complexity/cost for the Trusted OS

However, if you have a more limited use case or requirement (for example, you only need PSCI for ARMv7-A, or an authenticated boot for secure firmware, or you know exactly which Trusted OS you are going to use) then it might be easier to reuse relevant parts of Trusted Firmware and not initially aim to provide the full flexibility that the ARMv8-A solution does today.

sammoser, 2015-06-11:
Andrew, thanks for the response. I am an intern at Texas Instruments. We are looking to become familiar with ARM Trusted Firmware for use in our next generation parts as well as possibly adapt it for use in our current parts (which are ARMv7-A). We are just at the stage of investigating the suitability of Trusted Firmware for our current parts.

If we make a port of Trusted Firmware for ARMv7-A would ya'll be willing to review it for acceptance into the project?

sammoser, 2015-06-15:
Hello, I want to move forward on this and get started on an ARMv7-A port of Trusted Firmware, but not if it isn't something ya'll would be willing to take the time to review for possible acceptance into the project. Please let me know, thanks.

athoelke, 2015-06-16:
I think we'd be prepared to review some proposed changes, but I cannot promise that we would merge it upstream. At this stage we do not have a good enough idea of the complexity of the port or of the impact on the existing 64-bit frameworks, APIs, implementation and platform ports:
* the current "architecture" layering (the aarch64 directories) is incomplete, and we will only find out how incomplete it is by trying to port to 32-bit
* there are many private forks of this code base used to build firmware for ARMv8-A SoCs or provide Trusted OS integration. We do not arbitrarily modify the published interfaces used for platform ports and SPDs in order to protect the work done in our partner's code.
* without a significant user base, adding a potentially disruptive 32-bit to the code might not be justified given the impact on the existing users of this code.

danh-arm, 2015-10-07:
I'm going to close this specific thread since there has been no further comments for a while. The general topic of AArch32 support is still open, however...

closed 2015-10-07 by danh-arm