ARM-software/tf-issues #311: filesystem.cpio.gz

jbech-linaro, 2015-06-15:
I'm trying to step up ARM-TF and its dependencies to later versions for use in OP-TEE. I've been following the user-guide from scratch and there is one thing that I don't understand. In the past we have just put `filesystem.cpio.gz` directly in the root-folder of `Foundation_Platformpkg` (as described [here](https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/user-guide.md#prepare-ram-disk)) and we we have been good to go. The setup and how we start FVP is best described by [this](https://github.com/OP-TEE/optee_os/blob/master/scripts/setup_fvp_optee.sh#L425) line and [these](https://github.com/OP-TEE/optee_os/blob/master/scripts/setup_fvp_optee.sh#L449-L469) lines.

However, after stepping up to later versions it seems like the root FS couldn't be found:
```
...
VFS: Cannot open root device "(null)" or unknown-block(0,0): error -6
Please append a correct "root=" boot option; here are the available partitions:
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
...
---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
```

However, if I point to the root fs when configuring the kernel, everything works just fine (without doing any other changes)
```
     -> General setup
       -> Initial RAM filesystem and RAM disk (initramfs/initrd) support
```

Have something changed lately, or have I made any mistake when setting up my environment? Do we need to tell UEFI to provide the root FS now or, should it still automatically be found when put in the root of Foundation Models?

I'm using:
```
Foundation Models: FM000-KT-00035-r9p2-28rel0.tgz (with fvp-foundation-gicv2-psci.dtb)
ARM-TF: 789c220 Merge pull request #309 from soby-mathew/sm/fix_fvp_get_entry
edk2: 3346743 ArmPlatformPkg/Bds: Restore the setting of the FDT device path in the bo
linux: e208f5e mailbox: scpi: Fix initialisation of clk_set_value messages
```

// Regards Joakim

danh-arm, 2015-06-17:
Are you sure your EDK2 environment hasn't been changed to pass a kernel command line that points to a VirtioBlock FS instead of your RAM disk?

If you have followed the instructions to modify the EDK2 boot menu or platform file to point a VirtioBlock FS in step 3 here:
https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/user-guide.md#prepare-virtioblock
... then you will need to undo this, either by modifying the platform file or clearing out any persistent boot menu changes.

If that's not the cause, then I can't rule out the possibility of a change in a dependent component (EDK2 or kernel) that is causing the behaviour you're seeing. However, since we haven't moved the user guide up to the versions you're using, we can't confirm this.

danh-arm, 2015-09-10:
I'm closing this as things have moved on since this was raised. Please re-open if there's still a problem.

closed 2015-09-10 by danh-arm