ARM-software/tf-issues #313: Implemented always handling FIQ in EL3

virtualopensystems-plalov, 2015-07-14:
Hello,
We've extended the ArmTrustedFirmware to always handle FIQ in EL3, regardless execution world. This was done by implementing the INTR_TYPE_EL3.

Do you think this is a valuable for ATF and particularly to share the code?

achingupta, 2015-07-14:
Hi,

Thanks for expressing interest in sharing your work. You would need to sign the Contributor License Agreement in order to contribute to the project. Details are given in 
[contributing.md](https://github.com/ARM-software/arm-trusted-firmware/blob/master/contributing.md). 

Apart from the CLA, the relevance of your work depends upon the use case it targets. The main reason why `INTR_TYPE_EL3` is currently unsupported in ARM TF is that the secure world can only run in the GICv2 mode. GICv2 does not provide a hardware mechanism to distinguish between FIQs targetted to EL3 or S-EL1. So this becomes a messy problem for software to solve which involves coordination between EL3 and S-EL1 software which on real systems will be distinct entities.

If execution is in normal world then all FIQs will most likely be routed to EL3. The FIQ handler will have to figure out whether the interrupt is targetted to EL3 or S-EL1. In the latter case it will have to arrange for the interrupt to be taken in S-EL1. The EL3 handler must not change the state of this interrupt e.g. read the `GICC_IAR` to find the interrupt ID. By the time the S-EL1 handler will get a chance to handle the interrupt, another higher priority EL3 FIQ might turn up. So S-EL1 will have to hand control back to EL3. Even in the absence of such race conditions, an interface will have to be defined between EL3 and S-EL1 to do the handover.

If execution is in S-EL1 e.g. a Trusted OS, then an FIQ could be routed to EL3 or S-EL1. In the former case, S-EL1 software will have no control over masking exceptions. Every FIQ will preempt execution to EL3 and drop back to S-EL1 for FIQ handling. S-EL1 might be handling an interrupt when it is pre-empted. In the latter case, even EL3 FIQs will be routed to S-EL1 and will have to be handed back to EL3. Again, new interfaces will have to be designed.

There would be other scenarios as well all of which will contribute to software complexity. This is the reason why we treat S-EL1 as the owner of FIQs and do not worry about EL3 FIQs. GICv3 solves the above problem since it treats interrupts bound for EL3 differently from interrupts bound for S-EL1 and normal world. We do not support execution of secure world software with GICv3 support just yet.

How does your solution cater for the above constraints?

Thanks a lot,
Achin

achingupta, 2015-07-14:
Hi,

I have been made aware that VOS has recently submitted the CLA. Please ignore the initial bits of my last comment.

Thanks,
Achin

dukov777, 2015-07-14:
Hi,

Thanks for detailed clarification. My interpretation of INTR_TYPE was slightly different.
Our solution is to handle all FIQ in EL3(SCR_EL3.FIQ = 1) and then route them to S-EL1 - no extra communication from SEL1 to EL3.
Now, for me, is obvious that such scenario is doable with "routing model" and INTR_TYPE_EL3 is not needed.

Thanks,
Petar

virtualopensystems-plalov, 2015-07-14:
Hi,

Thanks for detailed clarification. My interpretation of INTR_TYPE was slightly different.
Our solution is to handle all FIQ in EL3(SCR_EL3.FIQ = 1) and then route them to S-EL1 - no extra communication from SEL1 to EL3.
Now, for me, is obvious that such scenario is doable with "routing model" and INTR_TYPE_EL3 is not needed.

Thanks,
Petar

closed 2015-07-14 by virtualopensystems-plalov