ARM-software/tf-issues #314: I'm not clear about bl2 secure state settings.

xlyu, 2015-07-22:
I saw we must make sure BL2 running at S_EL1,but the following code confuse me.  
function:

	bl1_plat_set_bl2_ep_info(){ 
	    SET_SECURITY_STATE(bl2_ep->h.attr, SECURE);//set bl2_ep->h.attr=0    
	    bl2_ep->spsr = SPSR_64(MODE_EL1, MODE_SP_ELX, DISABLE_ALL_EXCEPTIONS);//set cpu_mode=EL1t
	}
function:

	bl1_run_bl2(){ 
		... 
		if (GET_SECURITY_STATE(bl2_ep->h.attr) == NON_SECURE)//i think must comment it  
		change_security_state(GET_SECURITY_STATE(bl2_ep->h.attr))//if bl2_ep->h.attr =0, scr.ns=0,hint it's secure state;else scr.ns=1, no-secure
		...
	}
i think  `bl1_run_bl2` is not correct,  need comment if statement,change to:

	bl1_run_bl2(){  
		...
		//if (GET_SECURITY_STATE(bl2_ep->h.attr) == NON_SECURE)  
		change_security_state(GET_SECURITY_STATE(bl2_ep->h.attr))  
		...
	}  
am i right?

danh-arm, 2015-07-22:
You're right that there is a problem here, although it should not be causing any negative effect. BL1 always executes in secure state and always expects BL2 to run in secure state. So both of these lines should be removed. Also, we should add an `assert()` to check that `GET_SECURITY_STATE(bl2_ep->h.attr)` provided by the BL1 platform code is `SECURE` as expected.

It appears this anamoly is because a previous version of the code did have to handle transitions from both non-secure and secure states.

xlyu, 2015-07-23:
I got it. 
BL1 execute in EL3, and EL3 always is secure state(Like ARM-V7 monitor mode?) ,whether SCR.NS is 0 or 1.Do you think that SCR_EL3.NS bit default is zero, so  just need  to  check SCR_EL3.NS bit before executes BL2 ?

danh-arm, 2015-07-23:
Yes, SCP_EL3.NS will be 0 after reset; however, thinking ahead to when we add "firmware update" capability, it's possible that BL1 will transition to non-secure state before this code is executed. So on reflection, I think it's better to leave in this `change_security_state` line in case SCP_EL3.NS has previously been set to 1 (but still add the `assert` I suggested).

Thanks for spotting this.

xlyu, 2015-07-23:
I agree  with you adding  `assert` in development process.
thanks for your patience.

danh-arm, 2015-07-23:
I'm going to leave this open until we've merged a patch that fixes this.

closed 2015-09-04 by achingupta