ARM-software/tf-issues #317: Large memory support in xlat_tables

vicorjh, 2015-08-12:
The code within lib/aarch64/xlat_tables.c utilizes the assert routine as follows:

void init_xlat_tables(void)
{
	print_mmap();
	init_xlation_table(mmap, 0, l1_xlation_table, 1);
	tcr_ps_bits = calc_physical_addr_size_bits(max_pa);
	assert(max_va < ADDR_SPACE_SIZE);
}

There could be an issue when the "ADDR_SPACE_SIZE" is equal to or greater than 4 GB since the assert routine appears to take 32 bit arguments (and 4GB addresses are > 32 bits).

danh-arm, 2015-08-13:
From a pure C perspective you are right, but the AArch64 Procedure Call Standard (http://infocenter.arm.com/help/topic/com.arm.doc.ihi0055c/IHI0055C_beta_aapcs64.pdf) states that `unsigned long` is a double word (64-bits) on systems using the LP64 data model, which as far as I know is all current systems using Trusted Firmware. Let us know if you believe this to be a problem in practice.

Having said that, there is an open task to make the data type usage more consistent (#155) across the codebase. In this case, the data type should be changed to `unsigned long long`.

antonio-nino-diaz-arm, 2016-07-06:
This commit fixed it: https://github.com/ARM-software/arm-trusted-firmware/blob/3ca9928df202607e6d77e8b8a31a30ff4a934a4b/lib/xlat_tables/aarch64/xlat_tables.c#L88-L96

danh-arm, 2016-07-06:
Thanks - I'll close this then.

closed 2016-07-06 by danh-arm