ARM-software/tf-issues #318: Question:  mmap_region_t not defined

Mathieu-Brcm, 2015-08-17:
Hi, 
You may want to add:
#include <xlat_tables.h>
in:
include/plat/arm/common/plat_arm.h

This compile error is happening when non-arm platform use plat/arm/common files
Is this allowed?
Can plat/arm/common/xxx be compiled with plat/yyy/ platform code? or is plat/arm intended to be arm platform specific?

Thanks,
-Mathieu

danh-arm, 2015-08-18:
I think you're right. Is the compiler complaining about the undefined `mmap_region_t`?

Yes. Platform ports that want to be aligned with ARM development platforms (for example FVP and Juno) may use the functions/definitions in `include/plat/arm/common/` and the corresponding source files in `plat/arm/common/`. Such platforms are called "ARM standard platforms", even if they are not provided by ARM. Some of the other functions/definitions in `include/plat/arm/` may also be useful. This ARM platform code provides standard implementations for some of the required platform porting functions/definitions. However, using this code may require the platform port to implement additional platform porting functions/definitions. These additional functions/definitions are not documented and are subject to change without notice.

(I will add this information to the FAQ in the wiki).

Mathieu-Brcm, 2015-08-18:
Hi Dan,

Thank you for your reply.
Yes, the compiler complains about the mmap_region_t not defined.
I had to add the include in the plat_arm.h as indicated in my previous comment.

Regards,
-Mathieu

danh-arm, 2015-10-07:
Internal ref: http://jira.arm.com/browse/GENFW-1201

closed 2015-11-05 by achingupta