ARM-software/tf-issues #319: question reserve_mem error (fail to reserve memory)

Mathieu-Brcm, 2015-08-25:
Looking at the firmware-design.md (page 21) "Juno with BL3-2 in Trusted SRAM (default Option).

BL1 reserve memory for RW area (in arm_bl1_early_platform_setup)
BL1 reserve memory for BL2 (during loading)

Then this memory map is sent to BL2.
when BL2 wants to reserve space for BL3.1, it can not because the memory free area is before BL2 and the area between BL2 and BL1 is discarded and not available anymore.

I am missing something obviously?

athoelke, 2015-08-26:
You have a good point. Right now, we just happen to get "lucky". The positioning of BL2_BASE and BL1_RW_BASE means that when the BL2 memory is reserved the free space after BL2 is just larger than the free space before BL2, so the `reserve_mem()` function records the space after as free - so the use of this memory layout when loading BL31 is OK.

The use of the memory layouts dates from a time when we were considering a more dynamic load-time allocation of addresses which is still not implemented. At the moment, the ability to statically allocate addresses allows for re-use of RAM pages (BL1 data memory is later reused as BL31 data, etc), and avoids the need for larger position independent code or relocatable binaries.

You can see this in the somewhat ad-hoc way that BL2 determines the memory regions to use when loading each of the BL3x images. Clearly, a small change in memory layout right now might trigger the failure sequence that you describe above - and it might be problematic for other platforms if they choose to layout the BL images in a different way in SRAM.

Mathieu-Brcm, 2015-08-26:
Thanks for your response athoelke,
Moving BL2_BASE lower did the trick and created more free space between BL2 and BL1_RW_BASE than before BL2_BASE. This bigger area became the one used by reserve_mem() allowing BL31 to be loaded/reserved.

You may want to add a note on page 3 of the porting-guide.md document in the "BL2_BASE" section as a current limitation if user wants to follow ARM platform memory layout.
Or this may be more suitable to put in the firmware-design.md in the "memory layout on ARM development platform" section for non-ARM platform that want to align with ARM platform memory layout.

Thanks again,
-Mathieu

danh-arm, 2015-09-01:
I'm marking this a bug since we should at least document this strange behaviour better. In the longer term we should enable a better way of describing the image layouts that doesn't rely on internal framework behaviour.

danh-arm, 2015-10-07:
Internal ref: http://jira.arm.com/browse/GENFW-1200

closed 2016-02-19 by danh-arm