ARM-software/tf-issues #320: question: bl32 pointers manupulation should be ifdef out when BL32 is not used.

Mathieu-Brcm, 2015-08-26:
in plat/arm/common/arm_bl31_setup.c (line 167)
the following crash when BL32_BASE is not defined since bl32_ep_ingo->spsr is set to 0 by arm_get_spsr_for_bl32_entry() in arm_bl2_setup().

	bl32_image_ep_info = *from_bl2->bl32_ep_info;

it should be:
#ifdef BL32_BASE
	bl32_image_ep_info = *from_bl2->bl32_ep_info;
#endif

The proper address would be setup by BL2 (in bl2_plat_get_bl31_params()) only if there is a BL32 (decided at compile time).

I think an ifdef is required around the bl32_image_ep_info as indicated above in BL31 to be consistent with BL2. Or maybe a check for NULL but 0x0 could be a valid address so maybe not the best check.

Thanks,
-Mathieu

danh-arm, 2015-09-01:
You are right, although BL32_BASE is _always_ defined for ARM standard platforms so this problem should not arise in practice. I presume you are using some of the code in `plat/arm/common` but not all of it?

I think the simple #ifdef wrapper as you initially suggested is the best solution.

Mathieu-Brcm, 2015-09-01:
That is correct, I am using some plat/arm/common for a non-ARM platform because I want to stay aligned with ARM platform to re-use as much code as possible but not all of it (memory map differences).
Currently we don't have a BL32 (we will in future).
I think for SoC vendor, re-using plat/arm/common makes a lot of sense so far.

danh-arm, 2015-10-07:
Internal ref: http://jira.arm.com/browse/GENFW-1199

closed 2015-12-02 by danh-arm