ARM-software/tf-issues #322: Question: Where is best place to init DDR controller

Leo-Yan, 2015-09-09:
Hi,

The document: docs/porting-guide.md suggests to initialize DDR controller in the function bl1_plat_arch_setup(); This is a little confused, due BL1 is placed into ROM, and we usually need tune DDR timing and optimization.

So just want to check, if the DDR controller initialization should be placed into BL2 will be better than BL1?

achingupta, 2015-09-09:
Hi,

The porting-guide.md mentions the possibility of initialising the DDR in BL1 but does not mandate it. Maybe we should update it to leave the possibility of doing it in a later BL stage open. As long as the DDR controller is initialised and tuned before the memory is first used, it should be fine. On Juno, it is the SCP RAM firmware that does the DDR initialisation. This does not happen until BL2 transfers the firmware to the SCP. This is similar to your use case in a way.

hth,
Achin

danh-arm, 2015-09-09:
I think the porting guide is even more vague than that. It says:
"Platform-specific setup might include configuration of memory controllers ..."

This might be simple static memory controller configuration rather than DDR init. I certainly would not expect DDR init to occur in BL1 for the reasons Leo mentions.

Leo-Yan, 2015-09-09:
Thanks for explaining, it is make sense.

closed 2015-09-09 by danh-arm