ARM-software/tf-issues #324: Question : SCP firmware issue on JUNO board

virtualopensystems-kchappuis, 2015-09-15:
I have an issue with the recent version of ATF when I execute it on JUNO Board. 
As explained in ATF user guide https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/user-guide.md#arm-css-platform-specific-build-options, ATF doesn't support earlier SCP versions than v1.7.0 for compatibility reasons.
So with the BL30 version that I can download from ARM or Linaro website, I have this error message during the boot process:
INFO:    BL2: Initiating BL3-0 transfer to SCP                                                         
VERBOSE: Send info about the BL3-0 image to be transferred to SCP                                      
VERBOSE: Detecting SCP version incompatibility                                                         
ERROR:   Detected an incompatible version of the SCP firmware.                                         
ERROR:   Only versions from v1.7.0 onwards are supported.                                              
ERROR:   Please update the SCP firmware.                                                               
ERROR:   BL2: BL3-0 transfer failure                                                                   
ERROR:   Failure in platform-specific handling of BL3-0 image.                                         
ERROR:   Failed to load BL3-0 (-1)

So I would like to know if it's possible to download a SCP firmware version v1.7.0 or newer (BL 30) and where I can find it?

I already did some researches and I find a thread in ARM community about this issue http://community.arm.com/thread/8674 but after to test with the binary pointed in this topic I have always the same error message.

sandrine-bailleux, 2015-09-15:
Hello,

Have you also updated bl0.bin? Both the SCP Boot ROM (bl0.bin) and the SCP RAM firmware (bl30.bin inside the FIP) need to be updated to migrate to the new version of the SCPI protocol. Some modifications might be needed in the configuration files on your Juno board as well.

The simplest solution would be to install the full 'juno-latest-oe-uefi.zip' archive from Linaro's website. You can then replace the BL1 and FIP (that should include the new bl30.bin binary) images with your development versions.

Regards,
Sandrine

virtualopensystems-kchappuis, 2015-09-15:
Hi,

I did that you suggested in your last comment and I have no more issues with SCP firmware.
Thanks a lot for your help.

Regards,
Kevin

closed 2015-09-15 by virtualopensystems-kchappuis