ARM-software/tf-issues #327: `set_routing_model()` assumes both secure and non-secure contexts have been set

athoelke, 2015-10-06:


danh-arm, 2015-10-07:
Internal ref: http://jira.arm.com/browse/GENFW-1198

achingupta, 2015-10-07:
@athoelke 

Having looked at the code, I don't think the interrupt management framework (IMF) needs to call a context management API to set the right bits in the `SCR_EL3`. It is doing this on behalf of the caller i.e. the SPD. The data structures that the IMF maintains are sufficient for its purposes. 

There are two ways of setting the `SCR_EL3` in the context for a security state.
1.  Use the `cm_write_scr_el3_bit()` API if a context has already been initialised
2.  Use the `get_scr_el3_from_routing_model()` API if the context is being initialised to get the value of `SCR_EL3` and then set it directly in the context using `write_ctx_reg()`.

I don't think there is a need for the IMF to call 1. at all. The caller e.g. SPD would know whether the contexts have been initialised or not. It can use 2. to do what it is asking the IMF to do.

This means that all the SPDs will have to be changed as well. It is more complicated since there is a subtle change in semantics but seems to be the cleaner solution.

athoelke, 2015-10-07:
_Maybe_ that would have been better in the first place (though I am not entirely sure that having `set_routing_model()` ignore the existing cpu contexts makes sense).... but I would be reluctant to add source breaks like this without a stronger justification. As a side effect of fixing an unnecessary constraint we should not be breaking existing code.

closed 2015-11-19 by achingupta