ARM-software/tf-issues #329: Question : stick in "BL3-1: Initializing BL3-2" when excute atf on JUNO r1 board

waterchu, 2015-10-23:
I met a problem with the recent version of ATF when I executed it on JUNO r1 Board.
ATF commit: 4425448a01665aa16c36d32629934db694344f48
build parameters:
	ARM_TSP_RAM_LOCATION=dram	\
	DEBUG=1				\
	LOG_LEVEL=50			\
	PLAT=juno			\
	SPD=opteed			\
	BL30=${BL30} BL33=${BL33} BL32=${BL32}
BL0 BL30 downloaded from https://releases.linaro.org/members/arm/platforms/latest/
I got this message during the boot process:

VERBOSE: Reserved 0x6000 bytes (discarded 0x3000 bytes above)
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v1.1(debug):4425448
NOTICE:  BL1: Built : 10:37:54, Oct 22 2015
INFO:    BL1: RAM 0x403a000 - 0x403d000
INFO:    BL1: Loading BL2
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=1 at address 0x4017000
VERBOSE: Reserved 0x1c008 bytes (discarded 0x16000 bytes below)
INFO:    Image id=1 loaded: 0x4017000 - 0x401d008
VERBOSE: Reserved 0x6000 bytes (discarded 0x3000 bytes above)
NOTICE:  BL1: Booting BL2
INFO:    BL1: BL2 address = 0x4017000
INFO:    BL1: BL2 spsr = 0x3c5
VERBOSE: BL1: BL2 memory layout address = 0x401d008
NOTICE:  BL2: v1.1(debug):4425448
NOTICE:  BL2: Built : 10:37:54, Oct 22 2015
INFO:    BL2: Loading BL3-0
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=2 at address 0x4023000
INFO:    Skip reserving memory: 0x4023000 - 0x402e8a4
INFO:    Image id=2 loaded: 0x4023000 - 0x402e8a4
INFO:    BL2: Initiating BL3-0 transfer to SCP
VERBOSE: Send info about the BL3-0 image to be transferred to SCP
VERBOSE: Detecting SCP version incompatibility
VERBOSE: SCP version looks OK
VERBOSE: Transferring BL3-0 image to SCP
VERBOSE: Waiting for SCP to signal it is ready to go on
VERBOSE: Waiting for SCP_READY command...
VERBOSE: Sending response for SCP_READY command
INFO:    BL2: BL3-0 transferred to SCP
INFO:    Configuring TrustZone Controller
VERBOSE: TZC : Configuring region 0 (sec_attr=0x0, ns_devs=0x0)
VERBOSE: TZC : Configuring region (filters=0xf, region=1, ...
VERBOSE: TZC : ... base=0xff000000, top=0xffdfffff, ...
VERBOSE: TZC : ... sec_attr=0x3, ns_devs=0x0)
VERBOSE: TZC : Configuring region (filters=0xf, region=2, ...
VERBOSE: TZC : ... base=0x80000000, top=0xfeffffff, ...
VERBOSE: TZC : ... sec_attr=0x0, ns_devs=0x167f167f)
VERBOSE: TZC : Configuring region (filters=0xf, region=3, ...
VERBOSE: TZC : ... base=0x880000000, top=0x9ffffffff, ...
VERBOSE: TZC : ... sec_attr=0x0, ns_devs=0x167f167f)
INFO:    BL2: Loading BL3-1
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=3 at address 0x4023000
VERBOSE: Reserved 0xf008 bytes (discarded 0x5ff8 bytes below)
INFO:    Image id=3 loaded: 0x4023000 - 0x402c010
INFO:    BL2: Loading BL3-2
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=4 at address 0xff000000
VERBOSE: Reserved 0x27ab0 bytes (discarded 0x0 bytes below)
INFO:    Image id=4 loaded: 0xff000000 - 0xff027ab0
INFO:    BL2: Loading BL3-3
VERBOSE: Using Memmap
VERBOSE: FIP header looks OK.
VERBOSE: Using FIP
INFO:    Loading image id=5 at address 0xe0000000
VERBOSE: Reserved 0x1f000000 bytes (discarded 0x1efc8568 bytes above)
INFO:    Image id=5 loaded: 0xe0000000 - 0xe0037a98
NOTICE:  BL1: Booting BL3-1
INFO:    BL1: BL3-1 address = 0x4023000
INFO:    BL1: BL3-1 spsr = 0x3cd
INFO:    BL1: BL3-1 params address = 0x401d570
INFO:    BL1: BL3-1 plat params address = 0xf1e2d3c4b5a6978
NOTICE:  BL3-1: v1.1(debug):4425448
NOTICE:  BL3-1: Built : 10:37:54, Oct 22 2015
INFO:    BL3-1: Initializing runtime services
INFO:    BL3-1: Initializing BL3-2

Then the message has been stuck in the last line.please tell me how can I fix it ?

sandrine-bailleux, 2015-10-23:
Hello,

Could you please provide the following details:
* Which BL32 image did you put into the FIP image? Optee? In the build parameters you provided, what does the ${BL32} environment variable contain?
* If using Optee, where did you get it from? Did you build it yourself? Or did you use
some pre-built binary?
* Did you try with the TSP instead of Optee?
* Did you try without any BL32 image at all?

Regards,
Sandrine

waterchu, 2015-10-23:
hi,Sandrine

1.yes,it's optee.
the ${BL32} environment variable:
export BL32=$DST_OPTEE_OS/out/arm-plat-vexpress/core/tee.bin

2.I got the source code from https://github.com/OP-TEE/optee_os. commit:176c959dcbda2950a2a7b4853a7cf6bc571c1237
build parameters:
export PATH=$DST_AARCH32_GCC/bin:\$PATH
export CROSS_COMPILE=arm-linux-gnueabihf-
export PLATFORM=vexpress
export PLATFORM_FLAVOR=juno
export CFG_TEE_CORE_LOG_LEVEL=4
export DEBUG=1

no any pre-built binary else but bl0 and bl30

3.when I set build parameter SPD=tspd,The message has been also stuck in the line"INFO: BL3-1: Initializing BL3-2".
4.if SPD = none,atf goes well and passes to u-boot or UEFI.

Regards,
Water

vchong, 2015-10-23:
Mainline ARM-TF still doesn't have support to load an OP-TEE binary with a header? Ref:
https://github.com/OP-TEE/optee_os/issues/149
https://github.com/OP-TEE/optee_os/issues/197
https://github.com/OP-TEE/optee_os/issues/254
https://github.com/OP-TEE/optee_os/issues/400

Can you try https://github.com/linaro-swg/arm-trusted-firmware (optee_paged_armtf_v1.1 branch)? This fork is somewhat behind though, especially with the recent refactoring of mainline ARM-TF, and we're also having some issues lately with Juno - possibly some compatibility issues with recent bl0.bin and bl30.bin - which we're still trying to figure out.

ashutoshksingh, 2015-10-27:
Other option is to generate non-hashed optee_os image. Once the build is complete:
arm-linux-gnueabihf-objcopy -O binary out/arm-plat-vexpress/core/tee.elf out/arm-plat-vexpress/core/tee.bin
And build the trusted firmware again. (essentially regenerate fip.bin with new tee.bin).
This should temporarily solve your issue.

waterchu, 2015-11-17:
Thanks a lot.refer to vchong's advices,I fixed it.

closed 2015-11-17 by waterchu

vchong, 2015-11-17:
@waterchu Would you mind sharing your solution with the community?

waterchu, 2015-11-17:
I'd like to share，but give me a few days, it's a little busy recently.

waterchu, 2015-11-23:
Mainline ARM-TF still doesn't have support to load an OP-TEE binary with a header,so the output message window will stick in the line ‘Initializing BL3-2’,when I execut ARM-TF with OP-TEE on JUNO r1 Board.
Refer to https://github.com/linaro-swg/arm-trusted-firmware/commit/74efa34641672c420760a683896d6519116b0f5b#diff-458cae56af442eb352d33c1071fba500 and https://github.com/OP-TEE/optee_os/issues/400
I made a patch to ARM-TF(commit 84ab33e1e95500a3dfc776fe416b9d1a51288057).
[patch.txt](https://github.com/ARM-software/tf-issues/files/41251/patch.txt)

vchong, 2015-11-23:
@waterchu thanks for sharing!

XevenQC, 2017-11-03:
@waterchu, when you got stuck, how did you reflash the device? It cannot boot to fastboot when stuck at BL3-2 initalizing.