ARM-software/tf-issues #331: Question: wiki refers to AP_BLx but design doc to BLx same with SCP_BLxx.

Mathieu-Brcm, 2015-10-29:
There are some inconsistency in the naming between the wiki and design doc/code that might confuse people.

If the wiki provides the same information, it should refer to the doc already in git to avoid duplication and/or inconsistency.

Thanks,
-Mathieu

danh-arm, 2015-10-30:
Yes, there are some inconsistencies, in particular the wiki refers to the SCP firmware as SCP_BL2, whereas the code refers to this as BL3-0. The wiki is more up to date, and refers to the "old" names, which the code still uses.

The code will be updated with the new names before the next release. This was dependent on some other supporting patches that have only recently been merged.

danh-arm, 2015-12-22:
This has now been addressed in the v1.2 release

closed 2015-12-22 by danh-arm