ARM-software/tf-issues #333: Build fails when using Optee dispatcher

jcastillo-arm, 2015-11-10:
The Optee dispatcher makefile 'opteed.mk' sets NEED_BL32. This variable tells the main Makefile to build the Secure Payload from source if no BL32 image is specified in the command line. This works when Trusted Firmware is built using a single command (TF images + FIP) because BL32 points to the Optee binary, so the build system will not attempt to build a SP.

However, specifying BL32 is not mandatory when building the TF images. In that case, the main Makefile will attempt to build a SP and fail because opteed.mk does not include support for that.

FIP_ADD_IMG should be used instead of NEED_BL32 to indicate the build system that a BL32 binary should be included when building the FIP.

jcastillo-arm, 2015-11-10:
@jbech-linaro, @jenswi-linaro,
I am working on a patch and I will raise a PR once it is ready so you can provide feedback. Have you detected this issue on your build infrastructure?

jenswi-linaro, 2015-11-19:
@jcastillo-arm 
No we haven't seen this in our end, but then we use a fairly static configuration.

closed 2015-11-19 by achingupta