ARM-software/tf-issues #335: arm_configure_sys_timer() assumes optional registers to be implemented

sorenb-xlnx, 2015-11-14:
I was looking at getting support for Xilinx UltraScale+ Zynq MPSoC into ATF (I'll eventually create another issue for that). Overall, the gap between the Xilinx fork (https://github.com/Xilinx/arm-trusted-firmware) and upstream seems not too big. But one issue I ran into was in the common *arm_configure_sys_timer* function. It requires ARM_SYS_TIMCTL_BASE to be defined. I tried finding that interface in ZynqMP, but couldn't find it. And, IIUC, that interface is optional. Hence, ZynqMP does not seem to implement that optional memory-mapped timer interface. And ATF common code should probably not assume its existence unconditionally.

sorenb-xlnx, 2015-11-14:
I have a simple fix in https://github.com/Xilinx/arm-trusted-firmware/tree/tf-issues/335. I can create a PR if that is an acceptable solution.

achingupta, 2015-11-17:
Hi,

I will look into this today and get back. Apologies for the delay.

Thanks,
Achin

achingupta, 2015-11-17:
Hi,

I see that you are including `plat/arm/common/aarch64/arm_common.c` in the Xilinx port. Since ZynqMP does not implement the option system level memory mapped timer component, it does not define `ARM_SYS_TIMCTL_BASE`. This must result in a build failure at your end.

The issue is that `arm_common.c` is specific to ARM platforms all of which implement the system level timer. Even if we make the change you have proposed, the rest of `arm_common.c` could undergo changes due to requirements dictated by ARM platforms. You will always run the risk of breaking the Xilinx port due to this dependency. It would be better if you maintained a copy of this file instead of pulling it directly from `plat/arm/common`.

Please let me know if you need any clarifications,

Thanks,
Achin

sorenb-xlnx, 2015-11-18:
On Tue, 2015-11-17 at 11:27AM -0800, Achin Gupta wrote:
> Hi,
> 
> I see that you are including `plat/arm/common/aarch64/arm_common.c` in the Xilinx port. Since ZynqMP does not implement the option system level memory mapped timer component, it does not define `ARM_SYS_TIMCTL_BASE`. This must result in a build failure at your end.
> 

Right, that is the issue I see.

> The issue is that `arm_common.c` is specific to ARM platforms all of which implement the system level timer. Even if we make the change you have proposed, the rest of `arm_common.c` could undergo [...]

This seems odd to me. ZynqMP is an ARM platform. And most contents of
that file should be applicable to ZynqMP, IIUC. We also have a
implementation of the system timer (which I sincerely hope is compliant
with the generic timer architecture spec). But ZynqMP does not implement
the optional parts of the timer's memory mapped interface. While
duplicating the file would certainly be possible, shouldn't we make the
"common" code actually common? Depending on optional/implementation
defined features unconditionally doesn't strike me as something that
could be "common". And I'd also like to avoid syncing the actually
common parts of that file with some platform-specific duplicate.

	Sören

danh-arm, 2015-11-19:
I'm going to slightly disagree with Achin here - apologies in advance, Achin!

The code in `plat/arm/common` is for use by "ARM standard platforms". While this includes all ARM Limited (the company) development platforms, it may also include other companies' platforms that wish to align with the ARM Ltd development platforms. However, Achin is correct that ARM may make incompatible modifications to this code without warning. It's the customer's decision if they want to use `plat/arm/common/` directly and risk being broken, or use a copy of this code (and periodically resync if needed). 

So back to the problem in question. Your solution seems to rely on your specific platform not including arm_def.h or #undef'ing the ARM_SYS_TIMCTL_BASE. This is not a very robust or extensible solution. It would be better to add a new function in `plat_arm.h` called `plat_arm_configure_sys_timer()`, which is weakly defined in `arm_common.c` to call `arm_configure_sys_timer()`. The existing invocations of `arm_configure_sys_timer()` would need to change to `plat_arm_configure_sys_timer()`. Then you could provide a strong definition of `plat_arm_configure_sys_timer()` in your platform code, which does nothing (or something specific to your system timer implementation).

We would be willing to accept a pull request to this effect.

Regards

Dan.

sorenb-xlnx, 2015-11-19:
Hi Dan,

On Thu, 2015-11-19 at 10:00AM -0800, danh-arm wrote:
> I'm going to slightly disagree with Achin here - apologies in advance, Achin!
>
> The code in `plat/arm/common` is for use by "ARM standard platforms". While this includes all ARM Limited (the company) [...]

The arm_def.h header defines a lot of addresses that I suspect are very
platform-specific and it seemed impossible to re-use that file. E.g.
ARM_TRUSTED_SRAM_BASE is at a completely different address in our SoC.
And there are plenty more examples. So, ideally, these common files only
provide actually common symbols that can be used by all SoCs while
SoC-specifics would be provided by a platform-specific header.

Anyway, my approach was to have a platform-specific version of arm_def.h
(actually that is all in platform_def.h at the moment) that provides the
symbols that are needed, but use the common code as much as possible,
which resulted in the issue described here.

> 
> We would be willing to accept a pull request to this effect.

I was thinking along similar lines as you describe above. I'll work on
this.

	Thanks,
	Sören

sorenb-xlnx, 2015-11-19:
On Thu, 2015-11-19 at 10:00AM -0800, danh-arm wrote:
> I'm going to slightly disagree with Achin here - apologies in advance, Achin!
>
> The code in `plat/arm/common` is for use by "ARM standard platforms". While this includes all ARM Limited (the company) [...]

I was a little too fast. That would still break compilation when the
symbols like 'ARM_SYS_TIMCTL_BASE' do not exist, as it is the case for
ZynqMP.

	Sören

danh-arm, 2015-11-20:
> ... ideally, these common files only provide actually common symbols that can be used by all SoCs while SoC-specifics would be provided by a platform-specific header.

Common symbols that can be used by SoCs do not live in `plat/` at all. By definition, this is common code. The problem is that the amount of commonality between platforms is on a sliding scale, and what is common between your platform(s) and the ARM standard platform code is different for other customer platforms.

We can only realistically attempt to provide proper common platform abstractions for upstream platforms. Otherwise, it's impossible for us to know how relevant `arm_def.h` (and other) definitions are for other platforms. Are you planning to upstream your code?

Having said that, we can make tactical sharing improvements like this where the outcome is clear.

> That would still break compilation when the symbols like 'ARM_SYS_TIMCTL_BASE' do not exist, as it is the case for ZynqMP.
You should still include arm_def.h, even if you don't use (some of) the definitions in there. Several source files in `plat/arm/common` already do include `arm_def.h`.

sorenb-xlnx, 2015-11-20:
On Fri, 2015-11-20 at 06:37AM -0800, danh-arm wrote:
> Common symbols that can be used by SoCs do not live in `plat/` at all. By definition, this is common code. The problem is that the amount of commonality between platforms is on a sliding scale, and [...]

Yes, this issue is the first hurdle on that way. This branch has the
ZynqMP patch:
https://github.com/Xilinx/arm-trusted-firmware/tree/for-upstream (which
probably needs to be split into a few more commits).

> Having said that, we can make tactical sharing improvements like this where the outcome is clear.
>
> You should still include arm_def.h, even if you don't use (some of) the definitions in there. [...]

The memory map provided in arm_def.h is completely different from our
SoC. There is no way to share the current definitions for DRAM/SRAM
location & size, location of the ATF components, IRQ numbers,...
And again, ARM_SYS_TIMCTL_BASE is pointing to optional registers that don't
exist in ZynqMP. I'm fairly certain that defining that symbol without
backing it by any real HW is eventually gonna break something.
As I see it, if that file is truly supposed to be common, it would need
to be further split to factor out SoC(-family) specific parts into corresponding
platform/family headers.

	Thanks,
	Sören

danh-arm, 2015-12-22:
I'm going to close this as I think the original question has been answered. Any futher progress on this should be addressed by via pull requests to upstream the new platforms and to refactor the plat/arm/common code.

closed 2015-12-22 by danh-arm