ARM-software/tf-issues #336: Improve code related to memory attributes

sandrine-bailleux, 2015-12-03:
This issue lists a number of improvements that the generic code related to memory attributes could benefit from.

The generic code currently defines 3 types of memory:
* `DEVICE_nGnRnE`
* `DEVICE_nGnRE`
* normal memory

`include/lib/aarch64/arch.h` defines the memory attributes index field for `MAIR_EL3` as well as the memory attribute encodings corresponding to each one of these `AttrIndx` values in a translation table entry:
* `ATTR_SO_INDEX` == 0, which means that attribute index 0 in `MAIR_EL3` corresponds to Strongly-Ordered memory (the old terminology for `DEVICE_nGnRnE` memory). The corresponding memory attributes are encoded by `ATTR_SO` == `0x0`.
* Similarly, `ATTR_DEVICE_INDEX` == 1 and `ATTR_DEVICE` == `0x4`, which means that index 1 in `MAIR_EL3` corresponds to Device memory (the old terminology for `DEVICE_nGnRE` memory) and its attributes are encoded as `0x4`.
* Similarly, `ATTR_IWBWA_OWBWA_NTR_INDEX` == 2 and `ATTR_IWBWA_OWBWA_NTR` == `0xff`, which means that index 2 in `MAIR_EL3` corresponds to normal memory and its attributes are encoded as `0xff`.

There are a few issues with this code:
* It is not documented. We should add some comments, or maybe explain it in the Firmware Design doc.

* Some of the `#defines` names use the old ARM terminology. In the ARMv8-A architecture, "strongly-ordered" memory doesn't exist anymore, it's called "Device-nGnRnE" now. Similarly, "device" memory is too vague, as there several flavours of device memory. What used to be called just "device" memory is now more precisely "Device-nGnRE" memory. See section "E2.8.2 Device memory" in the ARMv8 ARM (ARM DDI 0487A.h).

* The "Device-nGnRE"/"Strongly-ordered" memory type is not actually used in the Trusted Firmware at the moment. The `ATTR_SO_INDEX` and `ATTR_SO` `#defines` should be removed to avoid any confusion. That will leave the `MAIR_EL3.Attr0` field unattributed so we might consider shifting the `MAIR_EL3.Attr1` and `MAIR_EL3.Attr2` values down, to make sure there are no "holes" in the register.

* The attribute indexes (1 for `DEVICE_nGnRE` and 2 for normal memory) should not be defined in `include/lib/aarch64/arch.h`, as these are not values defined by the architecture. Instead, they should be moved to `plat/common/aarch64/plat_common.c`, as they constitute the default memory types that most platforms would continue to use, but still allow platform ports to define alternative memory types.

sandrine-bailleux, 2016-01-05:
Internal ref: http://jira.arm.com/browse/GENFW-1281

antonio-nino-diaz-arm, 2018-06-21:
This has been fixed. For example, check https://github.com/ARM-software/arm-trusted-firmware/commit/04880e3fe66da49385ffe37583ead758c2961126

closed 2018-06-21 by antonio-nino-diaz-arm