ARM-software/tf-issues #338: General trusted firmware introduction or overview?

dirkbehme, 2015-12-15:
Is there a general introduction or overview for the trusted firmware?

There is quite a lot and good documentation how to use the trusted firmware, how the single parts work and how to port it to an other ARMv8-A platform.

Being familiar with ARMv7-A (Cortex-A9-SMP), the concept of using the trusted firmware on ARMv8-A seems quite different to what I'm familiar with.

What I somehow miss is a basic introduction of the basic idea of the trusted firmware:

* What's the problem with ARMv7-A the trusted firmware solves on ARMv8-A?
* Why is it split into several parts (BL1 / BL2 / BL31 / BL32)?
* What are the differences between these parts?
* Which part to use when?
* ...

Is this available anywhere?

Thanks

Dirk

danh-arm, 2015-12-15:
Hi Dirk

These kind of questions were addressed via a series of presentations at the Linaro Connect conferences over the last 2 years. Unfortunately, some of the older presentations do not seem be hosted on the Linaro website anymore. I will try to get these presentations uploaded to the Trusted Firmware wiki when I have time, although I'm currently very busy with the next release.

I'll leave this open until then.

Regards

Dan.

vchong, 2015-12-15:
Hi Dirk/Dan,

You should still be able to see most, if not all, those presentations here:
http://www.slideshare.net/search/slideshow?searchfrom=header&q=arm+trusted+firmware

Cheers

closed 2016-11-23 by danh-arm