ARM-software/tf-issues #340: Compilation error in psci_common.c with ENABLE_PLAT_COMPAT=0

sorenb-xlnx, 2015-12-29:
When ENABLE_PLAT_COMPAT is set to 0, *platform_get_core_pos()* is declared deprecated in *platform.h* using the macro *__warn_deprecated*, which is defined in *common_def.h*.
That header is not included in platform.h, resulting in compilation errors:
```
In file included from plat/common/aarch64/plat_psci_common.c:33:0:
include/plat/common/platform.h: In function 'platform_get_core_pos':
include/plat/common/platform.h:276:57: error: expected declaration specifiers before '__warn_deprecated'
unsigned int platform_get_core_pos(unsigned long mpidr) __warn_deprecated;
                                                        ^
```

sorenb-xlnx, 2016-01-04:
I changed my proposed fix as outlined in https://github.com/ARM-software/arm-trusted-firmware/pull/476. It is available in https://github.com/Xilinx/arm-trusted-firmware/tree/issues/340. Please, let me know if you would be willing to merge that, work on it on your side or need anything changed.

danh-arm, 2016-01-14:
Hi Soren. Thanks for your PR and sorry for the slow reply. We had a discussion our side about this and we think the right solution is to:

*    1st patch: Add the `__warn_deprecated` macro to `cdefs.h`, renaming it to `__deprecated` to be consistent with other macros there. From looking at the GCC docs, this support was added in version 3.1 so should be wrapped in `#if __GNUC_PREREQ__(3, 1)`. I'm not sure of the status of Intel compiler support for this but for now we can define it to be empty there. There is already a suitable block for the `__noinline` definition so this could be placed within that. Note, I changed my mind about creating a new `compiler_defs.h` file. I previously thought there might be some TF specific functionality to go in there but now I don't think there is.
*   (Optionally) upstream the same change to the FreeBSD project, https://github.com/freebsd/freebsd.
*   Patch 2: Remove the `__warn_deprecated` definition from common_def.h and the `#include <common_def.h>` statements from generic TF code (as you've already done in https://github.com/ARM-software/arm-trusted-firmware/pull/486). Also convert any usage of `__warn_deprecated` to `__deprecated`.
*   Patch 3: Replace all usages of __attribute__((unused))  and __attribute__((section(x))) across the codebase to __unused and __section(x) respectively. These are already defined in `cdefs.h`. This may require explicitly #include'ing `cdefs.h` in some places. 

No code should call `__attribute__` directly. From a quick search, I don't this this is being used in any other way in TF.

Let us know if you're happy to implement this.

Regards

Dan.

sorenb-xlnx, 2016-01-14:
Sounds good to me. I'll spend some time on it. It doesn't sound like
that much work.

	Thanks,
	Sören

sorenb-xlnx, 2016-01-15:
I updated the pull request as outlined above. I hope I didn't miss too much.

sorenb-xlnx, 2016-01-25:
This has been fixed in https://github.com/ARM-software/arm-trusted-firmware/pull/486

closed 2016-01-25 by sorenb-xlnx