ARM-software/tf-issues #341: Incorrect logic to enter or exit from DVM domain in CCN driver

achingupta, 2015-12-31:
`ccn_snoop_dvm_domain_common()` is responsible for providing a bitmap of HN-F and HN-I nodes in the interconnect. There is a request node (RN) corresponding to the master interface (e.g. cluster) that needs to be added or removed from the snoop/DVM domain. This request node is removed from or added to each HN-F or HN-I node present in the bitmap depending upon the type of domain. 

The above logic is incorrect when participation of a master interface in the DVM domain has to be managed. The request node should be removed from or added to the single Miscellaneous Node (MN) in the system instead of each HN-I node.

achingupta, 2015-12-31:
Internal ref. http://jira.arm.com/browse/GENFW-1309

danh-arm, 2016-02-26:
Fixed by https://github.com/ARM-software/arm-trusted-firmware/pull/516

closed 2016-02-26 by danh-arm