ARM-software/tf-issues #342: Perform CPU initialisations in BL1 during cold boot on secondary CPUs

sandrine-bailleux, 2016-01-06:
At the moment, the cold reset code in BL1 performs the following actions:

1. Warm/cold boot detection
   -> If it's a warm boot then jump to warm boot entrypoint.

2. Primary/Secondary CPU detection
   -> If it's a secondary CPU then jump to plat_secondary_cold_boot_setup().
      This function does not return.

3. CPU initialisations (i.e. reset_handler() & EL3 architectural inits)

This means that for a cold boot on a secondary CPU, step 3 is never reached.
This places the responsibility of performing the CPU initialisations on
plat_secondary_cold_boot_setup() (either in this function directly or indirectly
by jumping to some secondary CPU cold boot entry point that would take care of it),
when in fact it should be the responsibility of the generic code.

If we moved the CPU initialisations earlier, i.e. still after the warm/cold boot
detection but *before* the primary/secondary CPU detection then they would
be performed on all CPUs on a cold reset.

The ARM Trusted Firmware Reset Design will need to be updated accordingly.

danh-arm, 2016-01-06:
Note that current upstream `plat_secondary_cold_boot_setup()` implementations either:

a.  panic - i.e. secondary CPU cold boot is not supported.
b.  Immediately power down the secondary CPU, so the lack of CPU initialization is probably not an issue, or
c.  When `EL3_PAYLOAD_BASE` is defined, put the CPU into pen until a mailbox is populated during `bl1_plat_prepare_exit()`.

So the main concern here is the lack of secondary CPU initialization when bare metal EL3 payloads are used, although such payloads may take care of this intialization themselves.

Other platforms that have significant code in `plat_secondary_cold_boot_setup()` may be affected.

danh-arm, 2016-01-14:
Internal ref: http://jira.arm.com/browse/GENFW-1344

closed 2016-03-11 by danh-arm