ARM-software/tf-issues #348: Add support for MediaTek MT6795 chipset

coboy-chen, 2016-01-25:
This issue is intend to support for MediaTek's MT6795 chipset including platform code, related drivers, PSCI...etc.

opmss, 2016-04-07:
I am  using EVB mtk6795 development board. 
I downloaded it immediately and checkout 77a81df1e4a08ff6fb9b98aa9978cdeeb63f275a（commit-id：I4dd4398e83f61280e4bf1b9c32892a49edac7a83）, and made some changes to compatible with the Preloader parameter passing .
When USB cable is plugged, the system work well, because AP is not suspend at this time,  but when the USB cable is unplugged, the system will die and reboot for a moment.

The arm-trust-firmware log are:
INFO:  mt_on[0:3],entry 116c6c
INFO:  psci: core 0:3 ON
INFO:  psci: core 0:3 OFF
INFO:  psci: core 0:2 OFF
INFO:  psci: core 0:1 OFF
INFO:  psci: cluster 0 SUSPEND  (this is last print)

Do you have a similar problem? Could you please help me to solve the problem, I would appreciate very much.

Do you have stable arm-trust-firmware for mt6795 and Can you open them to share?

danh-arm, 2017-07-06:
This support is now upstream

closed 2017-07-06 by danh-arm