ARM-software/tf-issues #352: How can i get the bl33(Linux) work in secure mode when i don't have the bl32(secure os) enabled?

prime-zeng, 2016-01-27:
The reason is that:
Our soc has got some secure module registers which can only be accessed by secure mode cpu core.
1. when i have bl32(secure os), the secure module can be accessed by the module drivers in the secure os , that is working fine with atf.
2. when i don't have bl32(secure os), i need  to access the secure registers  with the module driver in bl33(Linux), but with atf, the bl33 is non-secure,so i got the problem.

danh-arm, 2016-02-01:
If you don't have a BL32(secure os) then you need some other code executing in the secure world that provides an SMC service to your Linux module driver. I would expect this to be implemented as an EL3 Runtime Service in your BL31 platform port, using the "SiP Service Call" range as specified in the SMC Calling Convention. See the documentation here:
https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/rt-svc-writers-guide.md

prime-zeng, 2016-02-02:
Got it, thank you. Indeed this is a way to resolve the problem, but why don't we design the atf like that:  Everything is allowed to work in the secure mode when i don't have to focus on secure, rather then fixed the Linux kernel running in non-secure mode. For the two reason:
1. In the past, we don't have the tee support and all the thing are done in the Linux kernel, but now we have to move something out even we don't focus on secure.
2. In the solution of no tee(bl32), your way to resolve the problem may have the performance problem, SMC service cost much more then direct access in the Linux kernel.

danh-arm, 2016-02-02:
Sorry, but you do not have a choice. In order to conform to the AArch64 Linux kernel booting interface, the firmware has to launch the kernel at either EL2 or Non-Secure EL1:
https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/tree/Documentation/arm64/booting.txt?id=refs/tags/v4.5-rc2#n149

If you do not want to conform to this, then we cannot help you.

closed 2016-02-22 by prime-zeng