ARM-software/tf-issues #355: Question: BL2 Loading BL3-1 in to Secure DRAM and not Secure SRAM

snappop, 2016-02-13:
Hi all,

We do not have enough spare SRAM to use for running EL3 in SRAM and want to load it in a TZC configured region of DRAM (DRAM was initialized in BL2).  We reclaim our SRAM for other functional purposes later.

Looking at the TBB I don't see any restriction on this, just on the BL1-->BL2 stage and earlier to be in Secure SRAM.

So my questions are:
1. Is it possible with the current code to change the base BL3-1 to load to Secure DRAM?
2. Is there any violation that I'm not aware of in TBB?  I didn't see it but maybe my version is old (oh boy version tracking!) ARM-DEN-0006B-2.

Thanks!
-Alan

danh-arm, 2016-02-15:
Hi Alan

1.  There are no Trusted Firmware restrictions on where you load your BL31 - you can configure this to be in TZC-secured DRAM in your platform port.

2.  I just checked the TBBR. The latest version I believe is ARM DEN0006C-1 (so I guess you should ask for a more recent version). BL31 must be in "Trusted RAM" but that may be Trusted SRAM or TZC-secured DRAM (or some other secure memory). In general, we (the TF team) would recommend using on-chip SRAM for EL3 code as it's more physically secure but we understand that this is not feasible on some platforms.

Regards

Dan.

snappop, 2016-02-16:
Thanks Dan,

That's what I figured but wanted to get some confirmation.

Yeah, my spec is really old... time to get the latest copy.

Thanks!
Alan

closed 2016-02-16 by snappop