ARM-software/tf-issues #356: Support for Xilinx UltraScale+ Zynq MPSOC

sorenb-xlnx, 2016-02-21:
We intend to upstream support for the [Zynq US+ MPSOC](http://www.xilinx.com/products/silicon-devices/soc/zynq-ultrascale-mpsoc.html). The SoC contains a quad A53 cluster running the ATF.
We currently maintain our fork with the platform port: https://github.com/xilinx/arm-trusted-firmware

Additionally to the base platform support, we have two console drivers, [cadence](https://github.com/Xilinx/arm-trusted-firmware/tree/master/drivers/cadence/uart) and [dcc](https://github.com/Xilinx/arm-trusted-firmware/tree/master/drivers/arm/dcc). I've seen that other platforms place console drivers within the platform files while we currently have them under *drivers*. Is there some guidance/preferred location for where to place drivers?

danh-arm, 2016-02-22:
Hi Sören

It's simply a question of how widely usuable you think your drivers are. If they are potentially reusable by other platforms that contain the same hardware device(s), they should be placed under `/drivers`. If there is something about the driver or hardware device that is specific to your platform, or you have low confidence in its shareability, then keep them under your platform port.

Regards

Dan.

danh-arm, 2016-02-29:
I think this has been answered.

closed 2016-02-29 by danh-arm