ARM-software/tf-issues #357: Question: why there is psci_do_pwrdown_cache_maintenance in the the function psci_suspend_to_pwrdown_start

prime-zeng, 2016-02-24:
psci_do_pwrdown_cache_maintenance will enable the dcache and invalidate dcache of the stack, i can't understand it in the function psci_suspend_to_pwrdown_start, can anyone explain?

soby-mathew, 2016-02-24:
I think the function you are referring to is `psci_do_pwrup_cache_maintenance` which will  invalidate the used stack and enable the dcache. It is invoked from `psci_cpu_suspend_finish` and `psci_cpu_on_finish`.

The `psci_do_pwrdown_cache_maintenance` will do the following with respect to Dcache:
* Disable the dcache
* Flush L1/L2 caches depending on whether it is CPU power down or cluster power down
* Flush the used stack and invalidate the free stack

This function needs to be invoked before powering down and hence it is invoked in 
`psci_suspend_to_pwrdown_start`.

Hope that clarifies your query.

prime-zeng, 2016-02-26:
got it, thank you so much.

closed 2016-02-26 by prime-zeng