ARM-software/tf-issues #358: Incorrect documentation regarding interrupt handlers

lumag, 2016-02-24:
In documentation it is written, that interrupt handler returns a pointer to the context which is used to restore state. However it seems that in reality the return value is ignored and context is restored by ```el3_exit``` using ```sp_el3```.

soby-mathew, 2016-02-24:
Yes, you are right. The context management in Trusted Firmware stores the current cpu_context pointer in sp_el3 and this allows us to ignore the return value from the interrupt handler. But this return value is useful if context management has to be done differently. The document can be modified to reflect this.

soby-mathew, 2016-02-24:
Internal ref: http://jira.arm.com/browse/GENFW-1402