ARM-software/tf-issues #359: TRUSTED_WATCHDOG_TIME_UID defined but not used in code

sbranden, 2016-02-24:
There is a certificate extension defined for TrustedWatchdogRefreshTime.  It is unused in the ATF code.  Will it be added to the cert_create tool and auth/image parser?

Is the plan to use this Watchdog time to specific the refresh time before jumping to a new image.  IE. If certificate extension present BL1 will turn on watchdog and set timeout before jumping to BL2.  BL2 will do the same when jumping to the next image, etc?

danh-arm, 2016-02-26:
The intention of this certficate extension is to allow overriding of the default watchdog time (256s) when passing control to the normal world during the firmware update process. This is an optional extension in the Non-Trusted Firmware Updater certificate, and we don't currently have a plan to implement it. We're happy to take a contribution to this effect, though.

Currently the TBBR does not specify the option of changing the watchdog refresh time during the normal Trusted Boot flow (BL1->BL2->BL3X). I agree it would make sense to also optionally use this extension for this purpose. To do that would mean adding this extension to the Trusted Boot Firmware certificate. I'll pass this on as feedback to the TBBR owners.

sbranden, 2016-03-01:
Yes, please add extension added as optional to all image transitions (or at least BL1->BL2).  I really only have a need of transitioning from ROM code to non-ROM at this time though.  If no other activity is going on in this area I will submit patches for feedback as we implement.  If others have implemented similar feature I'd be happy to follow what is available.

sbranden, 2016-04-08:
I have locally added the Watchdog to the certificate extension in the cert create tool.

Do you have any guidance on how/where it can be extracted from the certificate?
Should AUTH_PARAM_WATCHDOG be defined and used inside get_auth_param in mbedtls_x509_parser.

danh-arm, 2016-04-14:
We had an internal chat about this. You're right that a new AUTH_PARAM should be defined but this can be named something more generic like `AUTH_PARAM_PLATFORM` (since this is not used by common authentication code). The `mbedtls_x509_parser.c` implementation of `get_auth_param()` would simply return the certificate extension in the same way as `AUTH_PARAM_HASH` and `AUTH_PARAM_NV_CTR`. You will also need a new platform hook that is called just after `auth_mod_verify_img()` in `bl_common.c`. It would be passed the same args as `auth_mod_verify_img(). Your implementation of this platform hook should call `img_parser_get_auth_param()` using a `auth_param_desc_t` corresponding to the watchdog extension. The DER encoded extension will finally need parsing to get the watchdog value.

A bit fiddly I know but I hope that helps.

sbranden, 2016-04-14:
Thanks - this follows along the path I was going.  Right now I placed the extraction inline with auth_mod_verify_img.  I'll look into abstracting it to a platform hook.

closed 2022-07-22 by sbranden