ARM-software/tf-issues #360: TRUSTED_FIRMWARE_NVCOUNTER_OID certificate extension unused in code

sbranden, 2016-02-24:
TRUSTED_FIRMWARE_NVCOUNTER_OID is unused in ATF code.  Is the plan to implement rollback protection with this counter?

danh-arm, 2016-02-26:
The only reason we have not implemented this yet is the lack of a platform to test it on. The changes are relatively small. If it would help you, we can provide an untested implementation?

sbranden, 2016-03-01:
Yes, it would be great to provide an untested implementation.

danh-arm, 2016-03-01:
OK, I'll get this scheduled

sbranden, 2016-03-23:
Could you provide an example so I can see how to do this?

vwadekar, 2016-03-23:
If it makes sense for Tegra, I can get this tested on our platform too.

sbranden, 2016-03-23:
To begin with - I really just want to see how to extract certificate extensions at this point.

danh-arm, 2016-03-24:
This is coming soon (hopefully within a week or so).

alex-nemirovsky-ca, 2016-03-30:
curious how overflow state will be entered and if it will be handled differently than other "trackable" versions below the max. For example, if BL1 reads a BL2 cert with trusted minimal version set to 40. Do we accept it and set the counter to overflow state or do we reject it? I guess I am concerned about giving people a false sense of minimal version protection if they continue to issue certs incrementing minimal version counters in the certs beyond the max trackable version supported in HW (overflow condition).

danh-arm, 2016-03-30:
Hi Alex. It's the responsibility of platform code to check that the new value (from the cert) does not exceed the maximum value when updating the NV counter. It should fail if this is the case. In the code you will see soon, the ARM platforms do not support updating the NV counter at all so will always fail. On these platforms, the assumption is that the NV counter values in the certificates match the read-only values in the platform.

alex-nemirovsky-ca, 2016-03-30:
Thanks Dan.  Just to clarify.  Juno can't increment the FPGA NV Counters specified here?
http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dto0038a/CJHCGBED.html

danh-arm, 2016-03-30:
Correct: "For the purposes of the [Juno] development platform, the trusted and non-trusted counters are tied to the limit of increment"

alex-nemirovsky-ca, 2016-03-30:
Thanks Dan.  

So it sounds like we could provide code for Juno and could test the code on Juno but we at the expense of potentially "using up" the testing capability of physical test unit. Ince we reach the max incrementation count of a specific board, that board is no longer workable for any more incrementation requests.  In other words, we don't want to use up a Juno unit in order to test the code implemented for ATF.  Right?

Also, it seems that NV Counter section TBBR spec. closely aligns with Juno HW. i.e a minimum of 31 states plus an overflow state for trusted versions and a minimum of 223 states plus an overflow state for non-trusted.  In addition,TBBR also states what should happen if an authenticated cert contains a version number higher than what is found in the platform's NV counters. i.e. increment the platform's counter to match the cert.   Since this is in the TBBR spec., can I assume that the common code in the ATF is going to follow this TBBR description and automatically ask the platform to set the version number to match the higher number found in the cert?

I hope it's clear now why I was asking about what should happen when counter value exceeds the MAX count possible in my earlier post.  I think this type of decision may affect how the common ATF code is implemented and what functions are required of the platform. i.e.  return NV counter range.  Also, how do we get the HW platform counters into overflow state described in the TBBR spec? 

Sorry for all the questions, just trying to fill in some of the gaps between my understand of the platform capabilities, TBBR spec, and expectation what will be provided in the core logic in ATF driving calls to the platform NV counter interfaces.

danh-arm, 2016-03-31:
> Once we reach the max incrementation count of a specific board, that board is no longer workable for any more incrementation requests. Right?

No, the registers on Juno are read-only. They have the values of 31 and 223 for the Trusted and Non-Trusted NV counters respectively. Juno does not contain any One-Time-Programmable (OTP) memory.

There are 2 new platform functions; `plat_get_nv_ctr()` and `plat_set_nv_ctr()`. The latter always returns an error on ARM development platforms like Juno. Other platform implementation of this function should check for the MAX possible count as you suggest but this won't be done by the common code.

I think it's best to hold off on further discussion until the code is available.

danh-arm, 2016-03-31:
See https://github.com/ARM-software/arm-trusted-firmware/pull/572. Feel free to review.

sbranden, 2016-03-31:
thanks - I have only been looking at code in the master branch - did not know such code was available in the integration branch

alex-nemirovsky-ca, 2016-03-31:
Dan. 
 I was not aware of that code either. It' very helpful and answer most of my questions.  My only unanswered question is what we intended to do with the overflow state condition for trusted/non-trusted counters as requested in the TBBR spec. Hopefully the code when it's finally available will address that. 
Thanks again.

danh-arm, 2016-04-01:
> Hopefully the code when it's finally available will address that.

I don't understand. The code _is _available, and as I stated before, NV counter overflow must be handled by platform code. The ARM development platforms don't support NV counter update so cannot demonstrate this case.

sbranden, 2016-04-08:
I cherry-pick from the integration branch and tested - it appears to work well.  Good Job.

sbranden, 2016-04-14:
Code made it to master branch.  I think this issue can be closed.

closed 2016-04-18 by danh-arm