ARM-software/tf-issues #361: add --toc_flags <val> option to fip_create tool to populate flags field in TOC header

sbranden, 2016-02-24:
I propose to add a --toc_flags option to the fip_create tool.  This option allows the TOC header flags field to be populated with a value other than zero.

danh-arm, 2016-02-26:
Can you provide any information on how you intend to use this flags field? If we added an option to set this field without providing information on how its used, then there is a risk of future enhancements that use this field in a conflicting way.

It would probably better instead to add option(s) that set specific flags in this field for a specific purpose. As a minimum, we would need to provide documentation describing the use of these bits and which ones are reserved.

Currently, the only potential use for this field we have is 2 bits that specify the firmware encryption status, although we don't yet have any plans to implement this feature.

sbranden, 2016-03-01:
I was thinking of have a toc_flags option so all the flags could be specified in a generic manner without having to enhance the fiptool every time new flags are added.

I think the toc_flags may be platform specific.  Unless we wish to define some bits to have specific meanings and they have no meaning on other platforms (which I'm fine with).  Or, perhaps we define some of the bits for specific purposes and leave others reserved for platform usage?

We wish to use the toc_flags field to specify the PLL setting to speed up the BootROM execution.  Two bits would be used (00: bypass PLL, 01: 1/4 speed 10: 1/2 speed 11: full speed).  This allows the BootROM to speed up at the earliest possible time.

danh-arm, 2016-03-01:
There are 64 bits in the flags field. I think we could reserve the first 32 bits for common use and leave the remaining 32 bits for platform use. We could add an option to set platform bits in one go, but I wouldn't want to do the same for the common bits (for the reasons I stated before).

Given the sensitivity of your intented use-case, you may want to enable authentication / rollback protection on the ToC header (TBBR requirements R010_TBBR_TOC and R030_TBBR_TOC). This is not currently supported in the upstream code.

sbranden, 2016-03-01:
OK, we could add --platform_toc_flags option.
16-bits seems like enough to reserve for platform use to being with.  Other 16-bit could be reserved (set to 0) for some potential future use.  Current flags can then be reduced from 64 to 32 bits?

danh-arm, 2016-03-01:
OK, I guess we can start with 16 bits and expand later if needed. 64bits is what the current suggested ToC structure in the TBBR uses so I think we'd keep that. Perhaps we can have the following bit usage:

*   Bits 0-1: Firmware encryption status
*   Bits 2-31: Reserved
*   Bits 32-47: Platform defined
*   Bits 48-63: Reserved

Other fip_create options use dash chars, not underscores, so suggest --plat-toc-flags.

sbranden, 2016-03-23:
I can submit a patch shortly.  What is Firmware encryption status.  I thought this field is currently unused?

danh-arm, 2016-03-24:
>  I thought this field is currently unused?
It is, but when we do use it we want to use bits 0-1 for Firmware encryption status. This indicates whether the firmware is encrypted and whether it is using the Secret Symmetric Key (SSK) or Binding SSK (BSSK). See TBBR for more details.

closed 2016-08-09 by danh-arm