ARM-software/tf-issues #362: MMU improvements

lumag, 2016-02-24:
Improve the following aspects of MMU code:
- support larger page size
- support 4 level translation

This is required to support platforms with address space larger than 4Gb.

danh-arm, 2016-02-26:
Hi

Could you please elaborate on the use-case(s) you would like to enable here? We would normally expect the address space that a Trusted Firmware (TF) image needs to map in to be relatively small, and the existing translation table support to be sufficient. There is potentially a very significant RAM cost to the changes you propose.

In particular, it is usually better for a TF image to only map in the physical addresses it requires. If those physical addresses are spread across a large address space, it is possible to provide a non-linear address mapping so that the virtual address space is relatively small. If your virtual address space is still large, then presumably you are trying to map in large DRAM regions? Have you considered whether you need to map in all of those DRAM regions, and the security implications of mapping in those regions if the normal world can also access this memory?

Regards

Dan.

lumag, 2016-02-26:
@danh-arm 

We are using a ATF on a platform with full 48-bit address space. MMIO devices are spread through this space. So right now I have ```MAX_XLAT_TABLES``` set to 46. This covers secure region, part of non-secure RAM used for BL33 and several (lots) of MMIO devices that are set up or controlled from ATF. Using non-linear mapping is possible, but it results in uglier and more fragile code.

danh-arm, 2016-02-26:
> Using non-linear mapping is possible, but it results in uglier and more fragile code.
I agree it would be more complex to maintain if you have many sparse devices, but it would be a lot more memory efficient.

We've been giving some thought to #310. As part of that, one possible enhancement is to support simple allocation of virtual addresses while mapping in new regions. This would enable platforms to provide the same information as they do for `MAP_REGION_FLAT`, but end up with a contiguous virtual address space (except for the BL image memory regions).

Would that solve your use-case instead?

lumag, 2016-02-26:
@danh-arm In my case memory efficiency is not _that_ important, the code is anyway running from DRAM. #310 per se seems like and overkill for my use case. However your idea of 'simple allocation' seems appealing, if it's done in a logical way. Something like having 'mmio cookies' like done in Linux seems like a good approach.

lumag, 2016-02-26:
@danh-arm did some stats on MMU usage. I'm currently mapping 2Gb of device memory not including the PCI config space. And if I include PCI config space (necessary to setup some of devices), this goes up to 514 Gb. So fitting into 4Gb would definitely require dynamic mmapping/unmapping of device memory.

danh-arm, 2016-02-29:
I see the "simple allocation of VA" as a stepping stone to #310. I agree you don't need the full dynamic memory mapping solution. We'll have a look at the mmio cookies feature you refer to.

Although memory saving is not important for you, I'm reluctant to add bigger/deeper page table support since it might encourage poorer platform code in general. But I see there is a gap in features for you right now. I'll try to put a plan in place for this VA allocation support and get back to you. Until then, I'll put your PR on hold. Hope that's OK.

lumag, 2016-02-29:
@danh-arm I see your point. However I still don't quite understand, how can larger PA/VA support encourage bad platform code? Could you please shed some light on it, I'm worried if I'm ending up with some bad decisions in our platform code.

danh-arm, 2016-03-01:
All I'm saying is that if larger pagetable support is there, people will use it and potentially incur an unnecessary performance/ram usage cost. I'm not saying the code would be of poorer quality.

lumag, 2016-03-03:
@danh-arm What about just 4th level support? Because current code enforces us to use PA to VA translation, which ends up in quite a bad overall code design.

danh-arm, 2016-03-03:
I don't understand why you think address translation results in bad overall code design.

lumag, 2016-03-04:
@danh-arm Because with non 1:1 mapping I suddenly start getting a lot of per-device data (like base virtual addresses for registers). If you don't won't me to find those addresses in each function accessing the device, these base addresses should be stored in some variable. The only clean (!) and sane way to do this I can think of is to have these variables to be allocated dynamically (because there really can be different amount of devices of the same type on the board). I'd really like to have some kind of malloc-like memory allocation to be able to get rid of static globals, but I don't think this code is coming in the close future.

danh-arm, 2016-03-04:
OK, I'm starting to understand your use-case now; there are potentially some new requirements here. You are right we don't intend to add support for dynamic memory allocation in the near future but I see your need.

I will discuss this among our team here and let you know the outcome. Perhaps, the short term solution is to add the larger/deeper page table support you suggest.

athoelke, 2016-03-04:
I think I get Dan's concern: give people enough rope and they will hurt themselves. Making it easy to do the wrong thing with your memory map - "just map everything in case I need it" - could result in disabled security defenses.

In this situation, the trade-offs for the platforms that really need larger maps are also potential security concerns: code complexity stemming from non-identity mapping, runtime VA allocation, dynamic memory mapping.

I would suggest that we:
 1. start by introducing 4th-level translation table support, which is triggered by address space sizes of 40 bits or higher (and create a compile time error for >48 bits)
 2. tackle the "easy to do the wrong thing" issue with documentation regarding the security implications of getting your memory map just right

If we want to reduce unintentional use of 4-level translation, we could make this controlled by a separate build option, which works in conjunction with the `ADDRESS_SPACE_SIZE` (so only valid combinations are allowed).

We can then follow up by looking at the additional trade-offs involved in supporting 16KB or 64KB translation granules.

danh-arm, 2016-03-04:
Andrew captured my concerns but I agree we should add the 4 level page table support now, since we're not going to solve the other issues in this thread in the short term.

I notice the current PR for this allows the platform to define the page size and table depth (as well as the VA address size), but doesn't do any validation of these values so this would indeed give people "rope with which to hurt themselves". To start with, I agree we should get the xlat_tables code to detect  if `ADDRESS_SPACE_SIZE` is greater than 40bits and only then use 4th level tables. As part of this, I think it's going to be necessary to move some the macros in `arch.h` to `xlat_tables.h` (or .c) since they are specific to the xlat_tables library.

Dmitry - do you think you can update your PR to this effect?

I'm less concerned about unintentional use of 4th level translation so would prefer this not to be a build option.

I'm open to the idea of 16KB/64KB pages at some point, but calculating whether a 4th level table is needed would be more complex in this case, so I suggest we do it later.

lumag, 2016-03-04:
Sure, I will take a look at it during the Weekend.

What is a preferred process? Should I update an existing PR or open a new
one?

danh-arm, 2016-03-04:
I refer you to the wiki :o)
https://github.com/ARM-software/arm-trusted-firmware/wiki/Trusted-Firmware-Frequently-Asked-Questions#how-do-i-update-a-pull-request

As there are no comments, I don't mind if you update or create a new one.

closed 2016-09-02 by davidcunado-arm