ARM-software/tf-issues #363: SMC thread migration in TSP/TSPD

raghuncstate, 2016-02-26:
The current ATF implementation does not support thread migration for interrupted SMC's. Suppose we are servicing an SMC on core1 in TSP, and we get interrupted by a NS interrupt. ATF and TSP design dictates that NS world performs an SMC with TSP_FID_RESUME. This can be done by the NS world but TSP/TSPD makes the assumption that the TSP_FID_RESUME will be called on the same core it got interrupted on. This seems like a bug. 

Suppose we have core1 and core2 servicing SMC calls from the NS world on OS thread1 and OS thread2. Lets say core1 and core2 get interrupted during the SMC call and return to NS world with TSP_PREEMPTED. 
Before thread1 and thread2 call SMC TSP_FID_RESUME, Lets say they got swapped such that core1 now runs thread2 and core2 runs thread1. When the threads get around to executing TSP_FID_RESUME, different SMC's(different from the original SMC that the thread called) get resumed and they may get values from different SMC's. This seems like a security hole, whereby one thread is able to read another threads data. TSP/TSPD is allowing the circumvention of OS thread/process boundaries by returning data belonging to another thread.

athoelke, 2016-02-26:
Support of migration of secure tasks is certainly possible (and some Trusted OSes do that), but that was not a design requirement for the TSP.

The ARM Trusted Firmware code does not include the normal world OS driver code that is needed to make the TSPD operate correctly with regards to the issue you describe. For the TSPD, the normal world code should be executed from a thread that is locked to the CPU (at the time, either by using affinity or disabling pre-emption), and run in a loop calling `SMC_TSP_FID_RESUME` until completion before relaxing the CPU affinity. So it is within the control of the normal-world OS to prevent the cross-thread/process leakage of information. There are some Trusted OS normal-world drivers that do precisely this.

The TSP (Test Secure Payload) and TSPD (TSP Dispatcher) is just one example of a secure payload, its dispatcher and how fast/atomic and pre-emptible calls, and secure/non-secure interrupt handling could be implemented.

There are many other SP design possibilities and the TSP cannot demonstrate them all. Some production Trusted OS implementations provide task migration when a CPU is interrupted, some can run on any CPU but only on a single CPU at a time, some have a full OS scheduler and the SPD acts like a time-slicing hypervisor if both worlds are "ready to run" at the same time.

By the way, a task migrating TSP does not "solve" your identified problem - it just moves it around. How would the TSP/TSPD know which task to resume on a CPU? - the normal world must provide it with some form of thread or task identifier - and that means a malicious normal world could spoof a different thread's task ID.

raghuncstate, 2016-02-26:
Andrew,

Thanks for the response. Agreed TSP is an example implementation. Agreed that task migration support in TSPD does not solve the problem(and i realize now that there is no need to solve this problem in the secure world either). The simplest solution to prevent this would be for an OS to simply not allow preemption while making the SMC like you say.

closed 2016-02-29 by danh-arm