ARM-software/tf-issues #366: Rework GIC drivers to support multiple redistributors

lumag, 2016-03-03:
Our platform has several GIC instances  with single root distributor and several redistributors. However current GIC drivers assume that there is only one GICR register space. I ended up manually calling ```gicv3_rdistif_base_addrs_probe``` function manually from gic driver init function, but this is not optimal from my point of view.

soby-mathew, 2016-03-04:
Yes, this is a limitation of the existing GICv3 API. The discovery of multi socket re-distributers are not possible.

The current GICv3 driver expects the primary CPU to probe all the GIC re-distributors and populate it during cold boot. This needs to be changed to a  "per-cpu" re-distributor probe with mechanism for the platform to specify the GICR register space every time. 

I have created issue #369 for the same.

danh-arm, 2016-03-14:
I'm closing this given that Soby has created a task to implement this. Please comment there for any requirements/timeline, or if you intend to implement it.

closed 2016-03-14 by danh-arm