ARM-software/tf-issues #367: 64bit address map for SoC in BL31

jayannad, 2016-03-03:
Hi,

I am adding a new ARM64 platform port in arm trusted firmware. In out platform, on chip SoCs are mapped above 32 bit address space. Does BL31 support mapping devices / memory above 32 bit address space?
I am defining the address space size as 
#define ADDR_SPACE_SIZE                   (1ull << 48)

Getting the following error during compilation.
"arm-trusted-firmware/include/common/el3_common_macros.S:225:(.text.bl31_entrypoint+0x4c): relocation truncated to fit: R_AARCH64_ADR_PREL_LO21 against symbol `__RW_END__' defined in coherent_ram section in build/vvp1/debug/bl31/bl31.elf
./build/vvp1/debug/bl31/bl31_entrypoint.o: In function `bl31_entrypoint':"

Our BL31 image is mapped below 2GB space. I am using the toolchain gcc-linaro-aarch64-none-elf-4.9-2014.08_linux.

Thanks,
Jayanti

lumag, 2016-03-04:
@jayannad I believe you should be able to workaround/fix this (and similar) issues by replacing the offending line with the following one:
```asm
   ldr     x1, =__RW_END__
```

lumag, 2016-03-04:
@jayannad  However I'd suggest you to take look at the discussion at #362  and patches at https://github.com/ARM-software/arm-trusted-firmware/issues/535 .

danh-arm, 2016-03-04:
Hi Jayanti

Yes, TF should support an address space larger than 32 bits. ADDR_SPACE_SIZE specifies the maximum virtual address space. You can either increase that (as you have done) or use non-identity memory mappings to reduce the virtual address space size.

But it sounds you (and Dimitry) are reporting a bug. Are you running into relocation problems due to your BL31 memory exceeding 1MB (causing the `adr` instruction to attempt to find an address more than 1MB away from the current PC)?

Perhaps, we're going to have to review the code for large memory support. But if we start changing `adr` to `ldr` around the codebase, that will impact performance for everyone. So this would need to be configurable.

Up to now, we've been mostly considering how to make TF work efficiently in very RAM-constrained environments, but it looks like we need to think about providing some flexibility here.

Regards

Dan.

lumag, 2016-03-04:
They are running into this problem because with ADDR_SPACE_SIZE being large
enough L1 xlat table will grow expanding past the 1MB boundary. Note: this
error happens during compilation, not during linking. So there is no way to
check the BL size at that moment.

athoelke, 2016-03-04:
At present the upstream code assumes that it can map all the VA space with just 3 levels of page tables. For a 4GB address map that means the top level table only has 4 entries in it. This constraint means that you are limited to a 512GB address map (2^^39) with the current code. This will result in a 4KB top level translation table.

If you set the address size above this, the current code just makes that top table larger - which does not have the required effect in the CPU (but explains the linker error). A more specific compiler or linker error would help to diagnose this more quickly.

To support a larger VA space that 512GB, support for 4 level translation tables (or larger translation granules) is needed - Dmitry refers to his #362 issue and pull request regarding this.

danh-arm, 2016-03-04:
We have a patch in flight that adds a compile time assert for the implicit 39 bit VA space limit, which would have given a more useful compilation error, but that doesn't solve the fundamental problem.

Let's continue any further discussion on this in #362

jayannad, 2016-03-04:
Our BL31 memory limit is 64MB. I see this compilation error when the value
of ADDR_SPACE_SIZE is >= (1ull <<46).

Thanks,
Jayanthi

sandrine-bailleux-arm, 2016-03-07:
> We have a patch in flight that adds a compile time assert for the implicit 39 bit VA space limit, which would have given a more useful compilation error, but that doesn't solve the fundamental problem.

The patch Dan was referring to is there: https://github.com/ARM-software/arm-trusted-firmware/pull/542

danh-arm, 2016-04-18:
I'm closing this as a duplicate of #362

closed 2016-04-18 by danh-arm