ARM-software/tf-issues #370: Should the RO region BL1 be a MT_DEVICE instead of MT_MEMORY when ROM is used?

Mathieu-Brcm, 2016-03-04:
Hi ATF Team,

In the xla_tables.c mmap_desc() function, should there be a case for bootrom with (ATTR_DEVICE_INDEX | OSH) flag set without the XN bit set?

Or maybe we need another set of checks for MT_MEMORY case?

Regards,
-Mathieu

danh-arm, 2016-03-07:
Hi Mathieu

I'm not quite sure what you're asking for here. MT_MEMORY is the correct memory type for ROM; MT_DEVICE should be used for uncached device memory. Currently, we expect platforms to map all code, including ROM code, as RO (which results in the XN bit being unset). We expect platforms to map other BL image data as RW (which results in the XN bit being set); we assume the hardware will prevent any write transactions to ROM.

There is certainly a case for us providing independent control of the XN bit to allow platforms to map in ROM data (or RO data in other images) as RO and Execute-Never. Is that what you are asking for?

Regards

Dan.

Mathieu-Brcm, 2016-03-08:
Hi Dan,

I understand.
I will close this since the issue is likely on our side.

-Mathieu

closed 2016-03-08 by Mathieu-Brcm