ARM-software/tf-issues #371: IO memmap cannot be used to load images without the help of FIP

ljerry, 2016-03-07:
Hi,

In the case BL33 and possibly BL32 are stored at a hard-coded address in a memory-mapped storage, one could store them without the help of a FIP header.

In this case, platform should define
_static const struct plat_io_policy policies[] = {
	{
		BL33_IMAGE_NAME,
		&memmap_dev_handle,
		(uintptr_t)&bl33_block_spec,
		open_memmap
	}, {
		0, 0, 0
	}
};_
where
_static const io_block_spec_t bl33_block_spec = {
	.offset = BL33_IMAGE_BASE,
	.length = BL33_IMAGE_SIZE
};_

But load_image() fails due to the lack of size() function of IO memmap. Indeed io_size() returns IO_NOT_SUPPORTED.

It can be considered as a bug or at least a missing feature, bringing it is a slight enhancement.

With best regards
Jerry

closed 2016-03-15 by vwadekar

danh-arm, 2016-03-16:
Hmm, it's a bit weird that Varun managed to close this issue by pushing the relevant commit to his fork. I guess the GitHub "Fixes #XXX" syntax only looks for the commit being merged to "a" master branch, not necessarily the ARM-Software one. Perhaps a side effect of us using separate repos for issues and code.

vwadekar, 2016-03-16:
@danh-arm

That's funny.