ARM-software/tf-issues #372: Enable SCR_EL3.SIF bit

danh-arm, 2016-03-14:
The SCR_EL3.SIF (Secure Instruction Fetch) bit should be enabled in BL1 and BL31 common architectural setup code. When in secure state, this disables instruction fetches from
Non-secure memory.

We propose making this change non-configurably. Although this would break platforms that have secure world code executing instructions from non-secure memory, this is considered unlikely and would constitute a serious security risk. Such platforms should be corrected as a priority. Please comment if there any concerns about this.

danh-arm, 2016-03-14:
Internal ref: http://jira.arm.com/browse/GENFW-1435

vwadekar, 2016-03-14:
I agree. We already have this for our internal platforms. Not pushed
upstream yet. Thanks for taking care of this one.

vwadekar, 2016-03-28:
@danh-arm 

Hi Dan,

Any ongoing activity for this task?

closed 2016-04-08 by danh-arm