ARM-software/tf-issues #374: Provide mechanism for lower ELs to handle synchronous EAs locally when `SCR_EL3.EA` is set to 1

danh-arm, 2016-03-16:
Setting the `SCR_EL3.EA` bit allows EL3 code to handle asynchronous External Aborts (EAs, a.k.a. SError interrupts). However, this also affects how synchronous EAs are handled. To quote the ARMv8 ARM (ARM DDI 0487A.e):

> EA, bit [3] External Abort and SError Interrupt Routing.
*   0 When executing at Exception levels below EL3, External Aborts and SError Interrupts are not taken to EL3.
    In addition, when executing at EL3:
    *   SError Interrupts are not taken.
    *   External Aborts are taken to EL3.
*   1 When executing at any Exception level, External Aborts and SError Interrupts are taken to EL3.

Setting `SCR_EL3.EA` to 1 in Trusted Firmware currently prevents lower ELs from handling synchronous EAs locally, which can make debugging of synchronous EAs more difficult. Trusted Firmware should provide a mechanism for lower ELs to handle synchronous EAs.

Here are 2 possible solutions for this:

*   Provide an SMC API that allows lower ELs to register for synchronous EAs, instead of handling these directly in the lower EL exception vectors.
*   Modify the EL3 synchronous exception vectors so that after EL3 handling, a synchronous EA generated by a lower EL is redirected to that EL's synchronous exception vector.

guanhaohz, 2020-07-07:
@danh-arm 
Hi, is there any update for this issue?
I find Arm trusted firmware has macro "HANDLE_EA_EL3_FIRST", and it seems to be used for RAS framework.
So `SCR_EL3.EA` will only influence OS exception routing if you build with RAS feature right?

soby-mathew, 2020-07-17:
Hi @guanhaohz , 
Using SDEI[1] is the recommended way for handling these faults in lower EL when routed to EL3. This support is available in TF-A.
Please use TF-A mailing list as mentioned above for further queries.

[1] https://developer.arm.com/documentation/den0054/a/