ARM-software/tf-issues #375: Add new `plat_psci_ops_t` hook just before wfi in CPU suspend path

danh-arm, 2016-03-17:
The current `pwr_domain_suspend` hook in `plat_psci_ops_t` is called with locks acquired. Some platforms may want to take additional action after this point with the guarantee that the no global data is written to after this action is taken. For example, saving BL31 state to an alternative memory location during a system suspend operation.

This task is to create a new optional hook to enable this. Also consider related additional hooks (or re-use of the same hooks) in the other psci paths.

Documentation will also need updating.

soby-mathew, 2016-03-21:
Internal ref: http://jira.arm.com/browse/GENFW-1442

vwadekar, 2016-03-23:
Hello,

Any update on this activity?

danh-arm, 2016-03-23:
We intend to implement during next month, other priorities permitting.

vwadekar, 2016-03-23:
Will go ahead with a hacked version in our downstream code as the current
ETA would put us in a tight spot.

closed 2016-06-01 by danh-arm