ARM-software/tf-issues #376: eMMC isn't supported in ATF

hzhuang1, 2016-03-18:
In a lot of embedded platform, eMMC device is the only one storage device. So supporting eMMC stack is necessary in ATF.

vwadekar, 2016-03-18:
I am assuming that this driver would be necessary for lower level BLs - BL0/1/2. This seems like a worthy enhancement. Are you planning to add quirks similar to the linux driver? Quirks are a good way to allow platforms to override default behavior.

hzhuang1, 2016-03-19:
Yes, I will append similar quirks. For example, I only support CMD18 and CMD12 for read operation. But CMD23 and CMD18 should also be supported.

sbranden, 2016-03-23:
We have emmc fully functional in our non-upstreamed ATF version.  Lower level implementation is not clean.  I can contribute the io_emmc so we have compatible APIs?

hzhuang1, 2016-03-24:
I think you can submit your io_emmc first. Then Dan could make a decision.

sbranden, 2016-03-24:
Submission in github here:
https://github.com/sbranden/arm-trusted-firmware/tree/tf_issue_376

Please feel free to comment.

danh-arm, 2016-03-29:
I had a brief look at both pull requests (https://github.com/ARM-software/arm-trusted-firmware/pull/558 and https://github.com/ARM-software/arm-trusted-firmware/pull/566). Haojian's one appears to have a better abstracted IO driver, in that it introduces a new block device type instead of creating a specific eMMC block driver and re-using the memmap device type.

Scott - do you think you could review this (https://github.com/ARM-software/arm-trusted-firmware/pull/558) to see if it meets your needs?

Regarding the low level implementation, we (ARM) are not experts in this field and don't have a particularly strong interest in this driver. We will review it for general code quality but we need the TF community's help to produce a strong implementation.

sbranden, 2016-03-30:
I do not have the bandwidth to review such implementations in detail right now.  At first glance it will be adaptable or modifiable.   Our version works right now for our purposes.  I think we'll have to let Haojian do what he is doing and then fix/adapt as needed in the future.  It would be good if he could look at my commit to see what other features it has that are not already in his design.

danh-arm, 2016-07-05:
Closing this. The generic driver support was merged in https://github.com/ARM-software/arm-trusted-firmware/pull/597. The specific designware driver is still outstanding due to IP issues (https://github.com/ARM-software/arm-trusted-firmware/pull/599)

hzhuang1, 2017-01-20:
Reopen it.

hzhuang1, 2017-05-25:
Close since the fix is merged.

closed 2017-05-25 by hzhuang1