ARM-software/tf-issues #382: Add BL31 relocation support

sbranden, 2016-03-23:
Add support to make BL31 relocatable.  This can be enabled by defining BL31_ENABLE_RELOCATION.

danh-arm, 2016-03-24:
Can you explain what the use-case is here? If BL31 is relocated to its compiled address, then why can't the previous bootloader put it there in the first place? I don't like the idea of entrypoint code doing work before essential EL3 init.

athoelke, 2016-03-24:
My first reaction was also to misunderstand the use of "relocatable". This often implies that the image can execute from any memory address by updating any internal memory references before execution. Instead this allows the image to be loaded at any address, and will move itself to the compile-time run address before execution.

I can't think of a good name for this behaviour that differentiates it from the other use of "relocatable".

I can see that this has to run before the EL3 initialisation because the latter includes the CPU reset framework that has pointers to the reset functions which depend on the code running at the right address.

An implementation of this needs to protect against destructive overlapping copies (perhaps with an assertion or crash) in the case that the end of the loaded image region overlaps the start of the run region.

sbranden, 2016-03-25:
use case is when BL31 is used with some other bootloader that is not BL1/BL2.  These are compiled seperately and the memory address relation is not as hard coded as in the ATF build framework.

danh-arm, 2016-03-29:
Hi Scott. I'm still struggling to understand the use-case. What advantage is there to loading BL31 to one place and executing it from another? Is this to cope with some limitation/feature of the hardware? For example, does your BL1/BL2 load images to a fixed location, which have to be moved before loading a subsequent image to the same location? Or is there a security angle here?

If we do add something like this, I agree with Andrew that the term "relocatable" usually means something else. Perhaps `BL31_DIFFERENT_LOAD_RUN_ADDRESS`?

ScottShu, 2016-03-29:
Hi Dan,
   I’m afraid you've sent an email to the wrong person. ☺

Thanks and Best Regards,
Scott Shu
#23931

sbranden, 2016-03-30:
If you would like me to change the compile option to BL31_DIFFERENT_LOAD_RUN_ADDRESS I can do so.  If you don't want the patch I can maintain it locally.  Some developers find the feature useful for using BL31 with alternate bootloaders.  I use BL31 with BL1/BL2 so have no use for such feature either.

danh-arm, 2016-03-31:
I'm open to merging this; I just want to understand how it helps "some developers" first. Can you educate me?

sbranden, 2016-03-31:
Developer1 has existing bootloader support and wishes to start using BL31 but retain existing BL1/BL2 type code they have.  Developer2 uses all BL1/BL2/BL31 images.
BL31 link address does not equal load address of alternate bootloader.  Therefore same BL31 image cannot be used with 2 bootloaders without recompile.

danh-arm, 2016-04-01:
OK, I get it. The 2 different BL1/BL2 firmware solutions have different image layout strategies but you want to use the same BL31 for both without recompilation. I'll leave some review comments on the PR. If those are addressed, we can merge.

antonio-nino-diaz-arm, 2018-12-19:
Hasn't this been addressed in https://github.com/ARM-software/arm-trusted-firmware/pull/1644?

closed 2022-07-22 by sbranden