ARM-software/tf-issues #383: Boot redundancy support missing from BL1

sbranden, 2016-03-25:
The BootROM does not support boot redundancy.  This is necessary for fail safe image updates and/or flash corruption.

The boot redundancy needs to do the following:
1) Load from the default location of the boot source.
2) If the image is not found or image corrupt check next location (platform defined) (for example,NAND or SPI or USB flash this would be a compile time increment of offset in flash, for eMMC this should be Boot Partition 2)
3) Continue checking platform defined number of locations.  If no image found reset device.

I have not implemented this in BL1 at this time but it is a necessary feature we need to add.

danh-arm, 2016-03-29:
As far as I can see, there is nothing stopping BL1 platform code from implementing this (e.g. by overriding `bl1_plat_get_image_desc()` and `plat_error_handler()`). Perhaps there is a case for adding additional support in generic BL1 code but given that all policy would need to be in platform code, it's not clear to me what value would be left in generic code. Perhaps we need to see a working platform implementation first?

sbranden, 2016-03-30:
If there's no plan on supporting such feature we will develop internally.  It may or may not be worth our time to upstream.

alex-nemirovsky-ca, 2016-03-31:
Have similar need as Scott again on this one. +1

danh-arm, 2016-03-31:
OK, I'll leave this one open but we (ARM) don't currently have any plans to work on this.

sbranden, 2017-03-16:
Don't know how to link the pull request to this issue.

https://github.com/ARM-software/arm-trusted-firmware/pull/867

davidcunado-arm, 2017-03-16:
@sbranden add the following to the commit message:

Fixes ARM-software/tf-issues#383

antonio-nino-diaz-arm, 2018-12-19:
Fixed in https://github.com/ARM-software/arm-trusted-firmware/pull/867

closed 2018-12-19 by antonio-nino-diaz-arm