ARM-software/tf-issues #385: suggestion to retire confusing MMU #define MT_SECURE  or rename MT_NS

alex-nemirovsky-ca, 2016-04-01:
Hello,

As I understand it, ARM HW only as a NS  (aka MT_NS) enable bit in the MMU tables to force secure mode to do non-secure transactions to regions. MT_SECURE does nothing in reality and does not exist. 
This can lead to some confusion. For example, we can not elevate non-secure mode to use secure operations via MT_SECURE. 

I would like to suggest that we retire the definition MT_SECURE, and potentially also rename MT_NS to MT_FORCE_NS to clarify the function of this MMU NS bit. 

Thoughts?

danh-arm, 2016-04-01:
I really don't see a problem here. Specifying MT_SECURE in the translation table library is just another (and arguably clearer) way of saying "set NS = 0 in the memory attribute field of the translation table". It absolutely does have an effect when executing in the secure state. Yes this cannot be used by non-secure code to generate secure transactions, but that is simply a feature of the architecture. I propose no change.

alex-nemirovsky-ca, 2016-04-01:
>Dan wrote:
>  is just another (and arguably clearer) way of saying "set NS = 0 in the memory attribute field of the translation table".

As you are aware, the convention for setting up the table is to BIT OR in all of the MT_xxx attributes.
I am not sure how bit ORing MT_SECURE sets anything as you state above.  I hope you see why I state MT_SECURE does nothing, Leaving MT_SECURE out (i.e. 0 << 2)  completely out of the bitwise OR operation does the same thing; which is nothing.

danh-arm, 2016-04-01:
I don't buy your "wasted page table entries" argument. If secure world code wants to map in DRAM that is also accessible to the normal world, then it should map that in with the NS bit set, otherwise you'll end up with memory aliasing problems due to secure and non-secure transactions to the same memory. An extra page table entry is a small price to pay to avoid such problems.

I agree that setting MT_SECURE in the translation table library interface has the same effect as not providing any security setting, but I don't agree that it has no effect at all. If you find the MT_SECURE mnemonic confusing, then don't use it.

alex-nemirovsky-ca, 2016-04-01:
>Dan wrote:
> If secure world code wants to map in DRAM that is also accessible to the normal world, then it should map that in with the NS bit set, otherwise you'll end up with memory aliasing problems due to secure and non-secure transactions to the same memory.

so as you know the GIC block has both secure and non-secure registers access by both Secure and Non-secure worlds.   I believe the fall into the same page. Many TEE's allow GRP1 to be manipulated by NS worlds. What is the proper setting for this MMU mapping to avoid the memory aliases problem?

danh-arm, 2016-04-01:
Device memory is different; accesses to these registers are uncached so you won't get aliasing problems. The secure world will need to map this memory in with NS=0 if it wants to manipulate the secure registers. It doesn't matter what NS setting the normal world uses, but it's probably clearer to use NS=1.

alex-nemirovsky-ca, 2016-04-01:
Thanks Dan.

closed 2016-04-04 by danh-arm