ARM-software/tf-issues #386: Unhandled Exception in EL3

jayannad, 2016-04-04:
Hi,

I am adding a platform port in ARM trusted firmware. In our SOC, AArch32 is not supported at any excpetion level. Unhandled exception is observed while restoring the context from cm_prepare_el3_exit by calling el1_sysregs_context_restore.

Access to the following registers on AArch64 platform where AArch32 is not supported in EL1 is causing "Unhandled exception".
DACR32_EL2
IFSR32_EL2
FPEXC32_EL2
ARMv8 Architecture reference manual says that these registers are "UNDEFINED", If EL1 cannot use AArch32.

Thanks,
Jayanthi

danh-arm, 2016-04-05:
Yes, we don't currently test on AArch64-only platforms so this hasn't come up before. There should be a check for the EL1 register width supported by the platform before saving/restoring these registers.

Are you intending to provide a contribution to fix this?

danh-arm, 2016-04-05:
We had an internal discussion about this. There are some architectural nuances around when these registers should or should not be saved/restored. It's better if we fix this - I'll get this scheduled.

danh-arm, 2016-04-05:
Internal ref: http://jira.arm.com/browse/GENFW-1450

jayannad, 2016-04-05:
Yes, I am fixing this by checking bits[7..4] of ID_AA64PFR0_EL1 register.

athoelke, 2016-04-12:
There are a number of EL1 registers that are only used by AArch32, and I would suggest that it would be better to make this a build-time option, rather than a runtime check every time these registers are saved or restored. This would also allow the EL1 system register structure to be made smaller for systems where EL1 is AArch64 only.

_[I would need to check the ARMv8-A document for the full list of registers that can be elided on such systems.]_

jayannad, 2016-05-26:
The ARM architecture reference manual doesn't say that the following registers are not accessible from EL3 and I don't see any exception while accessing these registers from ARCH64 only platform.

SPSR_ABT
SPSR_UND
SPSR_IRQ
SPSR_FIQ

jayannad, 2016-05-27:
The following registers are RES0 if EL1 doesn't support execution in Arch32.
 SPSR_ABT
 SPSR_UND
 SPSR_IRQ
 SPSR_FIQ

danh-arm, 2016-05-27:
Yes, there may be no exception when accessing these SPSR_XXX registers on an AArch64-only platform, but there's no benefit in saving/restoring them. So they should be treated in the same way as the other registers (that do cause an exception).

closed 2016-06-06 by danh-arm