ARM-software/tf-issues #390: ARM model device-trees missing stdout-path property

leiflindholm, 2016-04-13:
Applicable to all the variants in /fdts/.

The stdout-path property in the chosen node removes the need to explicitly specify a default console.
A sufficient change would be:
-       chosen { };
+       chosen {
+               stdout-path = "serial0";
+       };

danh-arm, 2016-04-18:
Are you planning to raise a Pull Request with this change?

leiflindholm, 2016-04-18:
I hadn't planned to (mainly because I figured an update from wherever the canonical source for these are would drag this change in automatically).

danh-arm, 2016-04-18:
There's no canonical source for these (as far as I know) so there there's nowhere to merge from. So we rely on patches to keep these device trees relevant.