ARM-software/tf-issues #392: Stackpointer stored in psci_ns_context is not valid

jayannad, 2016-04-18:
clean_dcache_range is executed from bl31_entrypoint before calling el3_exit. But, the EL3 runtime stack (SP_EL0 ) which will be used for handling the next SMC call is updated in psci_ns_context from function el3_exit. When ERET is executed to return to lower exception level, the updated stack pointer is still in the cache and not updated in memory. The architecture initialization code at the next lower exception level would blindly invalidate the data cache by set and way. Due to this, when the SMC handler is called to handle the next SMC call in EL3 the stack pointer stored in psci_ns_context is not valid and causes synchronous exception at EL3.

achingupta, 2016-04-18:
Hi,

A couple of points regarding the issue you are facing. Could you please clarify?

1. Why is the lower EL blindly invalidating the data cache by set/way? On most ARMv8 processors caches are invalidated at reset. Even if they are not, then the platform reset handler should do this. If the lower EL is being initialised for the first time then why is the invalidation operation needed?

1. If at all there is a need to perform cache maintenance then this should be done by MVA. Use of set/way operation is recommended prior to power down and immediately after reset. In a platform with system caches, the set/way operations will not have any affect. The MVA operations are the only option.

1. If the lower EL is in the non-secure execution state then even a set/way operation will not impact the secure cache lines in any way.

It is a bit weird that a lower EL in the secure state can trigger a crash in EL3 by issuing a set/way operation. However, such usage seems incorrect to me. Cleaning the SP_EL0 prior to every ERET will have a significant performance penalty.

jayannad, 2016-04-18:
Hi Achin,

We are using the UEFI from Tianocore EDK2. After jumping to UEFI entry point from BL31, the PrePi module in ArmPlatformPkg invalidates the instruction and data cache by set and way. 
ArmPlatformPkg/PrePi/PrePi.c
   // Invalidate Data cache
  ArmInvalidateDataCache ();
  // Invalidate instruction cache
  ArmInvalidateInstructionCache ();

There is no security attribute associated with each cache line. So, the cache operation using set & way will impact all the cache lines.

Thanks,
Jayanthi

achingupta, 2016-04-19:
>    Link:

Hi Jayanthi,

> Hi Achin,
>
>    We are using the UEFI from Tianocore EDK2. After jumping to UEFI entry
>    point from BL31, the PrePi module in ArmPlatformPkg invalidates the
>    instruction and data cache by set and [...]

The instruction cache cannot be invalidated by set/way. I cannot really comment
on the rationale behind invalidating the D$ in EDK2. It might be there to cater
for legacy platforms where the SEC phase runs immediately after reset & the
caches are not invalidated by the hardware. It looks redundant on ARMv8
platforms. More below...

>
>    There is no security attribute associated with each cache line. So, the
>    cache operation using set & way will impact all the cache lines.

I am not following you here. On a ARMv8 processor with EL3 there will be a
secure and non-secure execution state. Each cache line has a security
attribute. This is not exposed to the programmer. A cache maintenance operation
from the non-secure state will not affect the secure cache lines. Please see
"Effects of virtualization and security on the cache maintenance instructions on
page D3-1631" in ARM DDI 0487A.i

UEFI will run in EL2 or EL1. So I still cannot understand how a cache
maintenance operation in UEFI is taking affect on a secure cache line.

Let me know what you think.

cheers,
Achin

danh-arm, 2016-04-20:
I wonder if EDK2 and Trusted Firmware has been integrated properly here? I believe the EDK2 code in ArmPlatformPkg/PrePi should only be used on platforms without Trusted Firmware (i.e. this code replaces Trusted Firmware). As Achin says, this code is redundant when Trusted Firmware is used.

It would be good to establish whether a lower EL has really induced a synchronous exception at EL3, since we should take measures to prevent this if possible. But as Achin again says, it's not clear how the lower EL could affect the cached secure world data.

jayannad, 2016-04-20:
Hi Achin,

Thanks for the clarification. Yes, each cache line has a security attribute. So, the DC invalidate from UEFI (EL2) will not impact the secure PA.

I am closing this issue as invalid.

Thanks,
Jayanthi

closed 2016-04-20 by jayannad